motioneye-project/motioneye #2043: DS220 + (Docker) the "motion detection" do not work...

zibous, 2021-02-28:
I installed ccrisan-motioneye on an **Ubuntu Server 20.10** and on a **DS220 + (Docker)** disk station. 

[image]

[image]

Everything runs fine on the Ubuntu server.

With the DS220 + (Docker) the "**motion detection**" does not work, even if I use the same camera settings as with the Ubuntu server.

- What solution is there ? 
- How can I upgrade the DS220 + Docker version to ccrisan-motioneye from 0.42 to 0.42.1?

starbasessd, 2021-02-28:
What initial install / run instructions did you use for Ubuntu vs DS220 + Docker?

zibous, 2021-02-28:
Switch to docker version `ccrisan/motioneye     dev-amd64           265779d6fdf2        2 months ago        494MB`: 
motionEye Version | 0.42.1
-- | --
Motion Version | 4.3.2
OS Version | Debian 10

Now all is working,  just a little bit of work because I had to reconfigure all of the cameras ...

closed 2021-02-28 by zibous