motioneye-project/motioneye #2044: Wiki - "(Install On Ubuntu (19.10 or earlier)"

gb-dursley, 2021-02-28:
The Wiki instructions "(Install On Ubuntu (19.10 or earlier)" are contradictory and confusing:-

"1. Install Ubuntu 20.04 or newer image (...32bit or 64bit)..."

but Ubuntu 20.x is 64bit ONLY, up-versioning to 20.x is NOT installing under "19.10 or earlier", and what about 19.3 LTS?

"7. Add an init script...
    Ubuntu 14.10...
    Ubuntu 15.04 or later..."

If Step 1. and release numbers are ignored (e.g. for 19.10 read 19.x) are the remaining steps accurate and reliable?

starbasessd, 2021-02-28:
"1. Install Ubuntu 20.04 or newer image (...32bit or 64bit)..." << Corrected, thanks for pointing it out.

Where is 19.3 LTS? Is it an Ubuntu fork?

"7. Add an init script...
Ubuntu 14.10...
Ubuntu 15.04 or later..."
Takes into consideration the switchover from init.d to systemd 

I will double-check the python2.7/pip2 in a bit, thanks again for pointing it out.

starbasessd, 2021-03-01:
FYI, while spinning up an Ubuntu 191.0 VM, had to change the repositories from http://us.archive... to http://old-releases... otherwise wouldn't install or update.

starbasessd, 2021-03-01:
making the changes to /etc/apt/sources.list in the previous comment, the instructions for 19.10 and earlier is confirmed working as of 3/1/2021.
The latest version of motion in the 19.10 repositories is 4.2.2

starbasessd, 2021-03-02:
Did my changes resolve your concerns?

closed 2021-03-05 by starbasessd