motioneye-project/motioneye #2046: Question about movie passthroug

Marc--, 2021-03-01:
Hi,

I'm using Motioneye since few years now, i'ts working great, but there is something i don't understand.

All my cameras are recording in "passthroug". Some of them have microphones. But in the recorded files, there is no audio.

I understand that there is no plan to add audio support in Motion or MotionEye, but... in passthroug, stream is untouched ?!? So why is there no audio ?

Or maybe it's an issue with my configuration ? I can see "netcam_highres ...." and "movie_passthroug on" in the ocnfig file, but not in the Web UI.

Thanks for your help !

Version info :
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Debian 10

starbasessd, 2021-03-01:
The VIDEO stream is saved as it is read, without conversion or compression.
If you read a mjpeg video stream from a camera, it saves that stream. If
you read a rtsp or onvif video stream, that is what is saved. There are
several hacks to add audio, but it's not 'built in'. Just video processing.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Marc--, 2021-03-01:
Hello,

Thanks for you quick reply ! I'm using RTSP on all my cameras. 

And i was thinking that audio was part of the "video" stream because when i open it with VLC for exemple, i can hear it.

I think it's a misunderstanding of my part !

starbasessd, 2021-03-01:
I have a Foscam 9900, the url I use is
rtsp://[ip_address]:88/videoMain but the only data motion/motionEye look
for is the video stream, even though in VLC I would get the audio, too (I
have it disabled on the cam, why eat the bandwidth?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2021-03-05 by starbasessd