motioneye-project/motioneye #2047: action buttons python issue

kmsgli, 2021-03-02:
Hey, I am having a problem using the action buttons to move my amcrest cameras.  The scripts work fine when i run them in the terminal but when I run them using the action buttons in motioneye I get a module not found error. Is this because arch linux uses python3 and motioneye is using python 2.7?

here is the error i get;
`Mar 02 14:12:37 archdesktop meyectl[23393]:  WARNING: up_4: command has finished with non-zero exit status: 1
Mar 02 14:12:37 archdesktop meyectl[23393]:  WARNING: up_4: Traceback (most recent call last):
Mar 02 14:12:37 archdesktop meyectl[23393]:  WARNING: up_4:   File "/etc/motioneye/up_4", line 3, in <module>
Mar 02 14:12:37 archdesktop meyectl[23393]:  WARNING: up_4:     from amcrest import AmcrestCamera
Mar 02 14:12:37 archdesktop meyectl[23393]:  WARNING: up_4: ModuleNotFoundError: No module named 'amcrest'
Mar 02 14:12:39 archdesktop meyectl[23393]:  WARNING: down_4: command has finished with non-zero exit status: 1
Mar 02 14:12:39 archdesktop meyectl[23393]:  WARNING: down_4: Traceback (most recent call last):
Mar 02 14:12:39 archdesktop meyectl[23393]:  WARNING: down_4:   File "/etc/motioneye/down_4", line 2, in <module>
Mar 02 14:12:39 archdesktop meyectl[23393]:  WARNING: down_4:     from amcrest import AmcrestCamera
Mar 02 14:12:39 archdesktop meyectl[23393]:  WARNING: down_4: ModuleNotFoundError: No module named 'amcrest'
`

Using pip2 to install amcrest module does not seem to solve my problem either. Any help is appreciated, thanks.

starbasessd, 2021-03-02:
If you are using motionEye, then make sure your script is calling /path to python/python3 /path to whatever command you are calling. Most flavors on *nix don't pass the environment paths to root or whatever user you are using to run motionEye.

kmsgli, 2021-03-02:
This is the script I am running i use the python3 shell 

`#!/usr/bin/python3

from amcrest import AmcrestCamera
import time
camera = AmcrestCamera('192.168.1.105', 80, 'admin', '**********').camera

#Move camera down 1s
camera.ptz_control_command(action="start", code="Up", arg1=0, arg2=0, arg3=0)
time.sleep(1)
camera.ptz_control_command(action="stop", code="Up", arg1=0, arg2=0, arg3=0)
`

starbasessd, 2021-03-02:
I am not a programmer.
Do you have both python 2.7 & python3 installed?
Which is default?
I am looking at a couple of other possible tests, let you know shortly...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kmsgli, 2021-03-02:
Python 3 is standard issue for Arch Linux but I thought by setting the environment to python3 in the script would solve that. when i run the script in the terminal with python2 I get the same error. I am not sure what is the best way to remedy this. It seems python2 does not have a amcrest module and setting the script to python 3 environment is not working.

starbasessd, 2021-03-02:
Try moving your script to another folder.
Create a new file with the same name and put in:

#!/bin/bash
/usr/bin/python3 /new path to the file/file

See if this works.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kmsgli, 2021-03-02:
Still same problem but I found some more info. when i run the command in the terminal via sudo it gives the same error s this because pip installed the amcrest module for the local user? should i use sudo pip to install the module or run as local user?

starbasessd, 2021-03-02:
Depends. If you want 'security' run as local user. If you don't care, do
the sudo pip install...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kmsgli, 2021-03-03:
> Depends. If you want 'security' run as local user. If you don't care, do the sudo pip install...
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

So sudo pip install did the trick for anyone else reading this.  

Thanks for all your help starbasessd, you have been very helpful and I greatly appreciate it.

closed 2021-03-03 by kmsgli