motioneye-project/motioneye #2048: Artifacts on a video stream

userosos, 2021-03-03:
motionEye Version | 0.42.1
-- | --
Motion Version | 4.3.2
OS Version | Debian 10
nuc5ppyh
Hi! On one of the cameras I have constant artifacts in the video stream. This triggers the recording in motion, as well as it is impossible to make out what is happening in the recording. See video file.
The same problem on a brand new video camera. If I view the video stream through VLC, there are no problems in the video stream.

starbasessd, 2021-03-03:
What camera? Make and model?
What camera URL? (rtsp://? http://? Something else?) If rtsp://, using UDP or TCP?
What resolution for the camera with artifact? Does it occur at lower resolutions?
How many cameras total? All Network?

starbasessd, 2021-03-03:
BTW No attached video?

userosos, 2021-03-03:
> 
> 
> What camera? Make and model?
> What camera URL? (rtsp://? http://? Something else?) If rtsp://, using UDP or TCP?
> What resolution for the camera with artifact? Does it occur at lower resolutions?
> How many cameras total? All Network?

It is an chine camera - IMIEYE HD 720P (rtsp, rtsp://IP:554/11), UDP or TCP - i don't know it, i set 800x600 and 1280x960
And Reolink E1 Pro (rtsp, rtsp://IP:554/h264Preview_01_main), UDP or TCP - i don't know it, 1080p and 800x600
[image]
I don't think it depends on the resolution. It's even more likely to be worse with 800x600 resolution.
All camera is network.

cap9qd, 2021-03-03:
To avoid sounding like a broken record I had more luck with the reolink
cameras running the rtmp; if you Google reolink and smearing its a big
topic. I don't have that particular camera but it is worth a try. That
picture looks exactly like the issues I had. Just Google rtmp, reolink and
motioneye to find my previous posts on it.

starbasessd, 2021-03-03:
For the IMIEYE camera try
rtsp://[ip_address]/11 (without the :554, per the iSpy website)
Is the UNIEYE camera having the issue or the Reolink?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

userosos, 2021-03-03:
> 
> For the IMIEYE camera try rtsp://[ip_address]/11 (without the :554, per the iSpy website) Is the UNIEYE camera having the issue or the Reolink?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I have the problem with both cameras.
Now i add the nem camera with rtsp://IP/11 and set UDP. I can see the pic:
[image]
I added udp - it work more good.

userosos, 2021-03-03:
In reolink i chaged to TCP and it has wery high latency and has Artifacts like in udp mode.

userosos, 2021-03-07:
I setting my reolink e1 pro camera used rtsp://192.168.89.146:554/h264Preview_01_main and TCP protocol and i can see very big latency and it incrementing everytime. If i use reolink app in my phone i can see 00/45/40 but in motionEye i can see 00/44/36. In motionEye i can see slowmotion stream the secs.

JamBalaya56562, 2026-11-08:
Closing this, as it comes from how this camera's RTSP stream reaches Motion rather than from motionEye: smearing over UDP and a growing delay over TCP are a known topic with Reolink cameras, as cap9qd mentioned. The camera's sub stream, a lower resolution or frame rate, or its RTMP stream helped others. If it still happens with a current Motion, the Motion project is the place to report it.

closed 2026-11-08 by JamBalaya56562