motioneye-project/motioneye #2049: Imou looc camera - not able to get stream

arkpoah, 2021-03-03:
Hello,

I have the stream url which works well in vlc, but when I add it in motioeye, I only get the no camera symbol.
My config is : 
```
netcam_url rtsp://192.168.1.76:554/cam/realmonitor?channel=1&subtype=1
netcam_userpass admin:password
```
the `frame Rate Dimmer` has a value greater than 0.
I also test to add `netcam_use_tcp` to `off`

I see nothing special in logs :/

starbasessd, 2021-03-03:
How are you adding the camera? Using the WebGUI or directly editing the .conf files?

arkpoah, 2021-03-03:
I added it using the WebGUI

starbasessd, 2021-03-03:
Did you try: 
rtsp://192.168.1.76/cam/realmonitor?channel=0&subtype=1
or other channels?

arkpoah, 2021-03-03:
yes, I get 404 errors in logs on other channels, only channel 1 is working with others rtsp clients.

arkpoah, 2021-03-04:
I just notice that when motioneye is running, my bandwidth received ~100KB/sec from my cam, so it is possible stream is working well and it is just an issue in visualization in webui ?

arkpoah, 2021-03-21:
I installed motioneye on my workstation with Manjaro, and it works well, so no problem with camera.
My issue is only when trying to display stream from motioneye installed on Ubuntu 20.04 LTS

Any idea wich package could be in cause?

rocket59, 2021-06-03:
You need to use this url:

rtsp://<ipaddress>:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif

However the lag on my setup (around 6s) makes the stream pretty much useless. If anyone has this working ok please post, otherwise this camera will be going into storage.

JamBalaya56562, 2026-11-10:
Closing, as a working URL was found: `rtsp://<ip>:554/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif`, as rocket59 posted.

closed 2026-11-10 by JamBalaya56562