motioneye-project/motioneye #2050: Can not adjust brightness

rindvieh1234, 2021-03-04:
Hi,
i just installed on a RP 3 A+ with this camera: https://www.amazon.de/gp/product/B07DVHT69B/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
It is not possible ti adjust brightness manual:

[image]

Any help please

starbasessd, 2021-03-04:
> It is not possible to adjust brightness manual:

That toggle is for Software Automatic Brightness when your camera doesn't
have it's own built in automatic brightness.
 It doesn't have any manual adjustments. (point your mouse at the little
blue question mark at the right end of the line)
If your camera is set to "Fast Network Camera" you should have manual
brightness adjustments
You can also use Extra Motion Options to adjust brightness using settings
from here:
https://motion-project.github.io/motion_config.html#Configuration_OptionsAlpha

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rindvieh1234, 2021-03-04:
Thanks for the help.

closed 2021-03-04 by starbasessd