motioneye-project/motioneye #2051: Timelapse files have wrong extension

af7567, 2021-03-05:
I'm assuming this was closed because there hasn't been any update on it for a long time, but the bug still exists. I have had a look into it and the problem is that the command to create a timelapse movie is using ffmpeg and the '-f avi' option so the file is always an avi.

mediafiles.py:695:
```
cmd += 'cat %(jpegs)s | ffmpeg -framerate %(framerate)s -f image2pipe -vcodec mjpeg -i - -vcodec %(codec)s ' \
               '-format %(format)s -b:v %(bitrate)s -qscale:v 0.1 -f avi %(tmp_filename)s'
```
But then the filename generated for the file uses the "Movie Format" set in the camera configuration, instead of always using avi

handlers.py:1269:
```
pretty_filename += '.' + mediafiles.FFMPEG_EXT_MAPPING.get(camera_config['movie_codec'], 'avi')
```
The Content-Type header is correctly set to tell the browser the data is video/x-msvideo though.

_Originally posted by @af7567 in https://github.com/ccrisan/motioneye/issues/1713#issuecomment-791638349_

af7567, 2021-04-25:
This problem should be fixed by pull request https://github.com/ccrisan/motioneye/pull/2110

zagrim, 2021-04-25:
Thanks for providing and implementing the fix. I see no reason why the amd64 build failed for the PR, doesn't seem to be related to the changes made at all (which FWIW look reasonable to me).

I've never actually used the timelapse feature, I think, and like you said in #1713 originally it seems like ME is creating a AVI containing MP4 (H264) video, but in my setup the file correctly has .avi extension. My "production" setup is a bit out of date, 0.41rc1, but I don't see any changes to those parts of code between that and 0.42.1, so I find it odd if you got .mp4 extension for the timelapses prior to the changes. :thinking:

af7567, 2021-04-25:
>  it seems like ME is creating a AVI containing MP4 (H264) video,
It is actually the other way round. In my camera settings I have "movie_codec mp4" so the timelapse code is creating a file with an mp4 extension. But the contents of this file is always AVI data and the Content-Type is set to AVI in the HTTP response.
So the generated timelapse is only MP4 by name, but everything else about it is AVI.

zagrim, 2021-04-25:
I also have `movie_codec mp4` (actually "mp4:h264_omx" but shouldn't matter), and the output file is named ".avi", the HTTP response from ME has `Content-Type: video/x-msvideo`, and the outputs of `file`, `mplayer` and `mediainfo` all report the (container) format as AVI with video stream being H264. So I guess for some reason my setup is not affected by what you describe. The code you changed had hard-coded file name extension ".avi" so I don't understand how it could possible have named the output file as ".mp4".

Anyway, if it works for you I guess it'll help also some other users, even if not all users might have been affected by this issue at all. Looking at the man page of `ffmpeg` it says the `-f` option you've removed shouldn't need to be used anyway so eliminating that might avoid any conflicting logic `ffmpeg` code might have related to that.

af7567, 2021-04-25:
The .avi extension was hardcoded in the tmp_filename var in mediafiles.py, this is used when ffmpeg generates the timelapse.
After the tmp file is created, handlers.py was using the variable pretty_filename which used the movie_codec option.

I guess different browsers handle the download differently. In Firefox I was downloading a file with an .mp4 extension but Firefox told me it was of type AVI. Some browsers might change the saved file extension depending on the Content-Type.

I was initially going to keep the -f option and choose the correct file format, but I also found that info in the man page and decided that ffmpeg was going to be better at choosing the format than me :)

MichaIng, 2022-03-19:
Solve in dev: #2110

closed 2022-03-19 by MichaIng