motioneye-project/motioneye #2052: Camera not detected on raspi zero w

Devyang-SHIFT, 2021-03-07:
hey
i have been working with this for more than a month now but getting the same again and again. Please help 

[image]

starbasessd, 2021-03-07:
What make & model of camera?
IP camera, CSI Camera, or USB camera?
What OS? 
Which install instructions?

Devyang-SHIFT, 2021-03-08:
its os is motineye os

the instructions follower are here :- [https://linustechtips.com/topic/927120-diy-wifi-cam-tutorial/](url)

my camera is "REES52® Raspberry PI Camera Module 5MP 1080p OV5647 Sensor HD Video Webcam Supports Night Vision for Raspberry Pi Model B/B+ A+ RPi 3/2/1/zero/zero W with FFC/FPC Cable"

its a pi camera

starbasessd, 2021-03-08:
The most common issue is the cable is installed wrong. There is a colored
piece (usually blue or black) on one side of the ribbon cable. that piece
needs to be away from the circuit boards at both ends.
Can you ssh or keyboard/monitor into the Pi?
If yes, run the command:
vcgencmd get_camera
and post the results.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-08:
Also, please close this, and open a new issue at:
https://github.com/ccrisan/motioneyeos/issues
and please answer all the questions it prompts.

Devyang-SHIFT, 2021-03-08:
the cable is correctly installed .The infrared lights turn on when i power the pi
i use motion eye os which does not support

closed 2021-03-08 by Devyang-SHIFT