motioneye-project/motioneye #2053: Same Videos on CAM2, 3 and 4

EmEiBee, 2021-03-07:
Hello from Germany...
... and sorry about my bad English ... and if I'm wrong here ...

I am running ME on an openmediavault (latest version, Debian 8.3.0-6) so far with a LAN camera. All good...
Today I tried to integrate a total of three identical external cameras. The string for this is "http://192.168.1.17 [X] /cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr= [user] & pwd = [pass]" where X = 1, 2 or 3
Camera 1 worked without any problems, camera 2 could still be integrated. But then both images produced the video from camera 2. So also camera 3 was integrated and now all images showed the video from camera 3.
I was unable to tell them apart.

Is this a known problem?

Greetz
Micha

[image]

[image]

[image]

starbasessd, 2021-03-07:
The IP addresses for each of the cameras needs to be for each of the cameras.
IOW
Camera 1 should be http://192.168.1.17...
Camera 2 should be http://192.168.1.18 ... (or whatever its IP address is
Camera 3 should be http://192.168.1.19 ... (or whatever its IP address is
and so on.

EmEiBee, 2021-03-07:
... hmmm ... The CAMs are *.171, *.172 and *.173..

Or do I misunderstand you? Are you saying that the cameras must be at least 10 IP addresses apart?

CRImier, 2021-03-07:
>>>Are you saying that the cameras must be at least 10 IP addresses apart?

no, certainly not.

You have to make sure that you didn't accidentally give same IP address to all cameras. You should check that in camera device settings, I presume.

EmEiBee, 2021-03-07:
Ok... The Cams are ok. I can use the "http: // 192.168.1.17 [X] /cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr= [user] & pwd = [pass]" link at any webbrowser and see every cam well. The Cams run since some month at the 3 different IPs and offer the snaps to the NAS ... So it can't be wrong configured also while using DHCP (ISC DHCPd Version 4.4.1).
You also can see the 3 different IPs of every cam in the screenshoots

starbasessd, 2021-03-07:
Sorry, I misunderstood what your [X] was indicating.
in (probably) /etc/motioneye folder:
You need to check for the camera-1.conf, camera-2.conf camera-3.conf, and camera-4.conf files, and make sure they are  correct.
make sure netcam_url is correct for each camera
make sure # @id is correct for each
in motion.conf make sure your cameras are listed correctly
Let us know.

EmEiBee, 2021-03-07:
... hmmm ... Looks all well. May take a look yourself in the attached ZIP (user/pass are removed and filled with [user:pass])
[ME.zip](https://github.com/ccrisan/motioneye/files/6097472/ME.zip)

starbasessd, 2021-03-07:
When you add a camera, only the URL should be added, then add username & password in the correct boxes.
http://192.168.1.173/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2
username [user]
password [pass]
Your zip files indicate you included the user & password with the URL.
[image]

starbasessd, 2021-03-07:
Camera 1 URL points to: netcam_url rtsp://192.168.1.188/stream1 and stream port 8081
Camera 2 URL points to: netcam_url http://192.168.1.171/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=[user]&pwd=[pass] and stream port 8082
Camera 3 points to: netcam_url http://192.168.1.172/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=[user]&pwd=[pass] but stream port also points to 8082 should be 8083
Camera 4 points to : netcam_url http://192.168.1.173/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=[user]&pwd=[pass] and stream port also points to 8082 should be 8084

EmEiBee, 2021-03-07:
1. I can't enter USER/PASS in the corresponding fields: that don't work. It must be a part of the URL. All other don't work.
2. I can't change the port of the CAM. So if I have to change the PORT of the cam also I can't use ME...

starbasessd, 2021-03-07:
You should be able to change the port in the camera-x.conf file directly,
then restart meyectl.
As to the username & password, can't help you further, Sounds like
something got corrupted.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

EmEiBee, 2021-03-07:
Ok, I have change the Ports in the config that is no twins... Now if I enable the 2rd CAM "an error occur. Refresh is needed". That also don't work. Refresh, restart, restart the whole server... nothing helps...

The USER/PASS is not a problem. That only works if the CAM directly ask about it (like an popup you know about .htaccess) Here these dumb cams need it as URL... So it works like that. I think that have nothing to do with the problem described before ...

EmEiBee, 2021-03-07:
Ok, first of all thanks for the help. That ultimately brought me to the solution.

What was not clear to me at first is what the meaning of the port specification in the Camera_X.conf is. I was of the opinion that this port is the one that is responsible for exchanging data with the camera itself.
After I recognized my mistake and also had to find out that the ME simply counts up the port with each additional camera, I was overcome by the simple realization that the ports around 8080 are often used by local services; here among others 8083-8086.
Since I did not find an option in ME to set the start area, I have now manually set the ports in the config to 8031 ++ and everything is good

starbasessd, 2021-03-07:
Good deal, glad to help.

closed 2021-03-07 by starbasessd