motioneye-project/motioneye #2055: Techage XM-IP608G-AI-20W: Empty reply from server

Krolik1988, 2021-03-08:
Hello, could some one helps me to connect this camera to Motion? I connected it to iSpy successfully,(using settings ONVIF http://xx.xx.xx.xx:8899/onvif/device_service) but in Motion with same settings I got "Empty reply from server". What I'm doing wrong? 

As example I can add it like:
rtsp://192.168.0.12:554/11
It see it as H264DVR 1.0RTSP/TCP Camera

But I will got only gray screen from it, no pic.

starbasessd, 2021-03-08:
did you try:
rtsp://ww.xx.yy.zz/channel=1_stream=0.sdp
???
Normally you remove the username & password from the URL.
on the iSpy site I did not see any special URLs for onvif.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Krolik1988, 2021-03-08:
Hi, I think I got it :) 
In Ispy link look likes:
rtsp://admin:password@192.168.0.12:554/user=admin_password=2emFQmmE_channel=0_stream=0.sdp?real_stream
In Motion it will be:
rtsp://192.168.0.12:554/user=admin_password=2emFQmmE_channel=0_stream=0.sdp?real_stream
User: admin
Pass: pass

closed 2021-03-08 by Krolik1988