motioneye-project/motioneye #2056: How to access UI and USB Webcam on stand-alone PC?

gb-dursley, 2021-03-08:
The Instructions (for motioneye under Linux) are written for a networked PC and one or more IP cameras.  In the past I've set-up such an installation with multiple PiCams running motioneyeOS.  
I'm now trying to set-up the most basic of installations of an old laptop (no WiFi) with a single **USB** webcam (in a bird-box).  
I seem to have installed motioneye under antiX 19.3 (Debian Buster, but sysvinit-based instead of systemd) but I can't find/work out how to access the browser-based GUI (the whole purpose of motioneye?) as neither PC or webcam have IP addresses!
I have a selection of webcams that all work perfectly with Cheese (I've only ever tried one at a time as this laptop only has 2 USB ports i.e. one USB controller only).
The reason for using motioneye are that I need to record any avian visits to this new birdbox.
Before I waste any more frustrating hours, can anyone please answer 3 simple questions:-
1. Is what I am attempting even possible?  
2. How do I access motioneye from my browser (what address)?
3. How do I configure motioneye to use a USB camera (instead of an IP camera)?
Thanks

starbasessd, 2021-03-08:
1. Is what I am attempting even possible?
        Yes, it is possible, but during setup you need to have internet
   access
        as it pulls various pieces and parts from internet resources.
        After set up, it can be stand alone.

   2. How do I access motioneye from my browser (what address)?
        http://localhost:8765 or http:127.0.0.1:8765

   3. How do I configure motioneye to use a USB camera (instead of an IP
   camera)?
        Select Local V4L2 camera

   Thanks

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-22:
If this answered your questions, please close.

gb-dursley, 2021-03-22:
Other things having taken priority over the last couple of weeks, I finally had a chance to try this yesterday.  Despite various error messages during the motionEye installation process, it all works perfectly without any further debugging.  On my initial login to the browser front-end, motioneye prompted me to install the USB camera.  Subsequently, swapping this out for the different, live, camera, motioneye recognised that as a different camera (installed as Camera 2). 
motioneye can run 2 or more USB webcams simultaneously.  I can't actually try this as the laptop in question has only 2 USB ports, one of which is needed for the mouse.  By the same token I can't test whether motioneye will run two identical webcams; I would expect to have to connect them to different USB controllers.
In the past, under Windows, I found it impossible to get software to distinguish between two different webcams, even where they were connected to different USB controllers.

closed 2021-03-22 by gb-dursley