motioneye-project/motioneye #2057: Credentials security issue

simiusstrabus, 2021-03-08:
When attempting to get a rtsp camera working, I had to look into some config files. All passwords are stored in clear text in motion.conf and any camera configs (when a network share is used for example). So if someone were to take your camera, they have your passwords. Yes, you should use different passwords for all websites, apps and what not, but most people don't. It would be a great addition to motion eye's security to store credentials encrypted in stead of plain text.

starbasessd, 2021-03-08:
Technically a Motion Project issue 
https://motion-project.github.io/
but tagged for possible enhancement/feature

JamBalaya56562, 2026-11-15:
motionEye's own user passwords are stored as argon2 hashes since 0.44.0, and its config files are kept readable by their owner only (0600). The camera, network share and SMTP passwords have to stay readable, because motion and mount need them in clear text, and a key stored on the same device wouldn't protect them from someone who takes it. Closing as not planned.

closed 2026-11-15 by JamBalaya56562