motioneye-project/motioneye #2058: AVCO M1060

hamdanfadi, 2021-03-10:
Hi,
I have an old Camera which is Avcom M1060W. I have succuessflly connect to it in iSpy, but now able to connect using motioneye.
The url that I have used is:

http://admin:123456@192.168.xx.xxx/videostream.cgi?

Can you please help me how to set it up?

Cheers,

starbasessd, 2021-03-10:
Should be:
http://192.168.xxx.yyy/videostream.asf
or maybe
http://192.168.xxx.yyy/videostream?resolution=VGA
or maybe one of the others found here:
https://www.ispyconnect.com/camera/avacom
with username and password in the appropriate boxes.

JamBalaya56562, 2026-11-10:
Closing, as this got no reply to the URLs suggested above (`/videostream.asf`, `/videostream.cgi` with the login in the username and password fields, or the ones listed by iSpy for Avacom). Please open a new issue with `motion.log` if the camera still does not connect with a current release.

closed 2026-11-10 by JamBalaya56562