motioneye-project/motioneye #2059: fast network camera feature request

BretG137, 2021-03-11:
I really appreciate all the hard work on this project and it works great for me.  I would just like to request that fast network camera gets added to motioneye similar to motioneyeos.  I have horrible wifi connection issues with motioneyeos on all my pi zero ws.  Running raspberry pi os and motioneye the connection is rock solid.  The only thing I am missing is the fast network camera feature.  Thanks!!

MagicPlants, 2021-08-15:
I second this, should be pretty easy right? Or is there an existing feature that does this that we're somehow overlooking? Thanks.

starbasessd, 2021-08-15:
I don't believe it's possible, because it (motioneyeOS only has enough OS stuff installed to get motion and motionEye running), and even then it disables most of motion and motioneye (using StreamEye) and most of the associated functionality which uses CPU and GPU cycles.

MagicPlants, 2021-08-15:
Not sure you read our question. We're asking if FNC can be added to Motioneye like it's added to MotioneyeOS. "motioneyeOS only has enough OS stuff installed to get motion and motioneye running" we're not talking about MotioneyeOS. I've been reading through the code and I'm thinking perhaps our solution is just to use streameye and forgo Motioneye altogether.

starbasessd, 2021-08-15:
The point of Fast Network Camera is to dedicate the Pi to streaming a video stream, and nothing else. You could add StreamEye to raspberry pi OS yourself, and use the 'functionality', just like you can use raspivid. It's here:
https://github.com/ccrisan/streameye
but to get the same 'improvement', you'd have to  turn off or disable most of the rest of the OS functionality.

JamBalaya56562, 2026-11-13:
As @ccrisan answered in #8 and #822, Fast Network Camera is a motionEyeOS feature, built on streamEye; on Raspberry Pi OS, streamEye (or another MJPEG streamer) can be run by hand and added to motionEye as a network camera, as @starbasessd described above. Closing, as it will not become part of motionEye itself.

closed 2026-11-13 by JamBalaya56562