motioneye-project/motioneye #2060: File storage on NAS

jartobec, 2021-03-11:
Hello

I am under ubuntu budgie 20.04 lts

I have two raspberry pi with motioneye which worked before and for 2/3 months upload / nas problem

Raspian and motioneye update performed

n ° 1 raspberry Raspbian GNU / Linux 9.13 (stretch)
motionEye Version 0.42.1
Motion Version 4.1.1
OS Raspbian Version 9.13

 
n ° 2 Raspbian GNU / Linux 10 (buster)
motionEye Version 0.42.1
Motion Version 4.2.2
OS Version Raspbian 10

All photos are directed to a NAS

n ° 1 directs photos to the NAS

n ° 2 **does not direct** photos to the NAS

The settings in file storage are the same (except the location) and the test services are ok = Accessing the Upload Service Succeeded!

I tested n ° 2 on google photos = ok / with filezilla = ok / with my **NAS = not ok**

the command: `journalctl -u motioneye`

hereafter an extraction of the return

`mars 10 16:14:16 raspberrypi meyectl[359]:    ERROR: failed to upload file "/var/lib/motioneye/Camera1/2021-03-10/16-14-04.jpg" with service ftp:
mars 10 16:14:16 raspberrypi meyectl[359]: Traceback (most recent call last):
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
mars 10 16:14:16 raspberrypi meyectl[359]:     service.upload_file(target_dir, filename, camera_name)
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 103, in upload_file
mars 10 16:14:16 raspberrypi meyectl[359]:     self.upload_data(rel_filename, mime_type, data, ctime, camera_name)
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 875, in upload_data
mars 10 16:14:16 raspberrypi meyectl[359]:     conn = self._get_conn()
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 912, in _get_conn
mars 10 16:14:16 raspberrypi meyectl[359]:     self._conn.login(self._username or 'anonymous', self._password)
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/lib/python2.7/ftplib.py", line 394, in login
mars 10 16:14:16 raspberrypi meyectl[359]:     resp = self.sendcmd('USER ' + user)
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/lib/python2.7/ftplib.py", line 251, in sendcmd
mars 10 16:14:16 raspberrypi meyectl[359]:     return self.getresp()
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/lib/python2.7/ftplib.py", line 217, in getresp
mars 10 16:14:16 raspberrypi meyectl[359]:     resp = self.getmultiline()
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/lib/python2.7/ftplib.py", line 203, in getmultiline
mars 10 16:14:16 raspberrypi meyectl[359]:     line = self.getline()
mars 10 16:14:16 raspberrypi meyectl[359]:   File "/usr/lib/python2.7/ftplib.py", line 193, in getline
mars 10 16:14:16 raspberrypi meyectl[359]:     if not line: raise EOFError
mars 10 16:14:16 raspberrypi meyectl[359]: EOFError
`

The error is well detected!

Do you have a solution to eliminate this dysfunction?

thanks for advance
have a good day

starbasessd, 2021-03-11:
Curious:
3 different OSs listed
Ubuntu Budgie. Where the FTP service is located?
Raspbian 9.13
Raspbian 10.
2 different levels of motion
4.1.1 & 4.2.2

Are you running full desktop or Lite on the Raspberry Pis?
Which Pis?
What is the Ubuntu on?
If the FTP server is 'Mars", try changing to the IP address for the FTP
server URL
There were some major changes in how networking was handled between Stretch
(9.x) and Buster (10.x)

Do you use 'anonymous' for uploading? Check username & password.
If you use a complex password, make sure you aren't using characters like
$, #, and a few others. THe different versions of motion and how they
handle things like that are concerning.
meyectl[359]: self._conn.login(self._username or 'anonymous',
self._password) mars 10 16:14:16 raspberrypi

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jartobec, 2021-03-12:
Thank you for the feedback
I misspoke

My two raspberries are my local network. (Lite version)
I drive them from ubuntu budgie with VNC.

The server is a Freebox
[image]
with your explanations  (There were some major changes in how networking was handled between Stretch
(9.x) and Buster (10.x)) I have to modify the access and the rights

I will get back to you as soon as possible

Thank you

jartobec, 2021-03-19:
Hello,
 After more in-depth investigations and changes to the appropriate parameters: **functioning OK** ;)
Thank you

closed 2021-03-19 by jartobec