motioneye-project/motioneye #2061: A Mauve Picture

Torchy61, 2021-03-11:
How do I get into the configuration file for altering the Hue, Contrast etc.  Even in bright sunshine the photo or video is mauve.
A square piece of blue plastic came with the camera. Is this anything to do with it.

starbasessd, 2021-03-11:
Yeah, it's a NoIR camera (No Infrared Filter), and the plastic is an IR Filter.
The 'mauve' color is due to not using the IR Filter.
You really can't adjust the hue, contrast, etc. to adjust that setting...

Torchy61, 2021-03-11:
Ooh Many thanks for that.
So I basically just attach the blue filter to the front of the lens?

starbasessd, 2021-03-11:
That should work.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2021-03-11:
If you are using fast network mode you can set the exposure mode to "gray
world" and it will help quite a bit if you have any issue using the ir
filter film. I think there may be a way to do it with the "extra motion
options" field but I have not done it before.

cap9qd, 2021-03-11:
This is the post:
https://github.com/ccrisan/motioneye/issues/1370#issuecomment-527471418

cap9qd, 2021-03-11:
And this for fast network camera mode:
https://github.com/ccrisan/motioneye/issues/1758#issuecomment-633257606

Torchy61, 2021-03-11:
Thank you 

> That should work.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Thank you, that is most helpful. I'll try the filter in the morning.

Torchy61, 2021-03-11:
> And this for fast network camera mode: [#1758 (comment)](https://github.com/ccrisan/motioneye/issues/1758#issuecomment-633257606)
> […](#)

Thank you.

starbasessd, 2021-03-11:
Fast Network Camera is a motionEyeOS only feature, and this is the
motionEye support. If they are using the motionEyeOS, they need to switch
to that github...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cap9qd, 2021-03-11:
Oh. Sorry.

starbasessd, 2021-03-11:
@cap9qd no worries. I'm not always sure if there should be 2 support sites, myself, sometimes.  I do wish there was a better way to separate the issues related to other OSs vs motionEyeOS, but unless they are obviously talking about one on the wrong other site, I usually just try to go with the flow...

Torchy61, 2021-03-12:
Hi guys,
Attaching the blue plastic to the front of the lens doesn't seem to make a difference, it just makes the picture a little more purple.
Is this because I haven't left the filter on long enough, and needs time to adjust?

Torchy61, 2021-03-12:
[image]
[image]

Torchy61, 2021-03-13:
I am using a Raspberry Pi Zero W, with a Raspberry Pi NoIR camera V2, and just using the Motion Eye OS on its own.
I am very new and this is my first project and learning as I go along.

I made sure the camera ribbon was attached in the correct way. I read the instructions very carefully to ensure the software was installed correctly. I did not use the blue squared filter at this point.

"starbasessd" very helpfully pointed out that the filter needed to be attached. I tried this and the results are above. There are no settings I can use to adjust the colours (like in MS Photo app). 

I am in the process of using another SD card, loaded with Raspbian OS and adding Motion Eye to this. I get the feeling the Pi Camera might work better under its own OS. Maybe I'm incorrect. Motion Eye does seem to be installing well, so I might go back to the original method of just using Motion Eye OS on its own.

So I have this Mauve, purple coloured haze in both video and still footage. It has to be the camera and not the software. 

Do you think the camera is at fault?

closed 2021-03-13 by Torchy61