motioneye-project/motioneye #2062: Raspberry motioneye and pi cam (strange color pink)

bend94, 2021-03-12:
Hi 
I have motioneye on raspbian installed with a picam (included infrared)
my issue is that the color is pink !
do you know why ?
regards

functionalprototype, 2021-03-13:
Can you post an image?  I had a camera that was "NO IR", that is, no IR filter, but it was mistakenly marked as a normal camera.

bend94, 2021-03-15:
Hi
thanks for your reply. i did many tests. What i find the best is to add that line in the /boot/config.txt

awb_auto_is_greyworld=1

Now, the colors are not perfect but i don't have this purple color
Thanks 
best regards

benjamin-salchow, 2021-08-11:
Thank you. I had the same Issue and the `awb_auto_is_greyworld=1` in /boot/config.txt fixed that. I guess this issue can be closed :-)

JamBalaya56562, 2026-10-29:
As found above, `awb_auto_is_greyworld=1` in `/boot/config.txt` fixes the colours of NoIR cameras with the legacy camera stack. With the current libcamera stack, the NoIR tuning file is selected as described here: https://github.com/motioneye-project/motioneye/wiki/Troubleshooting#raspberry-pi-noir-cameras. Closing as completed.

closed 2026-10-29 by JamBalaya56562