motioneye-project/motioneye #2063: Eufy Cam

hamdanfadi, 2021-03-13:
Hi there,
I just successfully added Eufy 2C cam, but I can see any video stream. Only showing Grey Screen. Any suggestion? 

Cheers,

Fadi

hamdanfadi, 2021-03-13:
I have made it working, but after some times 2-3 min it become not available

starbasessd, 2021-03-13:
You didn't give any setup info to begin diagnostics:
Type of board? Pi, PC, something else
What OS version?
What connections for the camera & computer? Wifi or ethernet?
What URL are you using for the camera ? rtsp://[ip_address]/live2 ?
TCP or UDP?

hamdanfadi, 2021-03-13:
Hi there,
Here are my answers :

Type of board? Pi, PC, something else
What OS version? Pi 4
What connections for the camera & computer? Wifi or ethernet? Ethernet
What URL are you using for the camera ? rtsp://[ip_address]/live2 ? Yes this is what I’m using
TCP or UDP? TCP

Cheese,

Hamdan, Fadi

24B Carrick Glen Avenue<x-apple-data-detectors://2/1>, Flat Bush, Auckland 2019<x-apple-data-detectors://2/1>

New Zealand

Mobile: +64 210 660088<tel:+64%20210%20660088>

Email: ***@***.******@***.***>

starbasessd, 2021-03-13:
MotionEyeOS? Which version? There is a date in the filename
motion/motionEye on another OS? Which OS (and it's version)
Try re-setting up the camera with UDP.

hamdanfadi, 2021-03-13:
Hi,

I have the latest version of motioneye and I have tried UDP. Still same issue. It works for 1-2 minutes then stopped working

Cheers,

starbasessd, 2021-03-13:
You aren't answering the question(s)
Motion/motionEye or motionEyeOS. They are not the same. motion/motionEye runs on another OS. This is the github you are in. motionEyeOS is self-contained. Its support forum & github are at https://github.com/ccrisan/motioneyeos/issues
If motionEyeOS, what is the filename you used to install from?
If motion/motionEye which set of instructions to install did you use?
These are very specific questions because each has different issues, can cause different problems, and have different fixes.
I haven't even begun to ask questions about the camera. There are a few out there that have 'power saving mode'. I couldn't find an online manual, if you have one, please link it here.

hamdanfadi, 2021-03-13:
Hi there,
Here are the information that you requested:

motionEye Version       0.42.1
Motion Version  4.2.2+gitUNKNOWN
OS Version      motionEyeOS 20200606
motionEye Version       0.42.1
Motion Version  4.2.2+gitUNKNOWN
OS Version

starbasessd, 2021-03-13:
Someon sent me a link for the manual. It does not stream continuously when there is no motion. It does not stream continuously unless the app is connected. I goes into a sleep or power saver mode because it is battery powered. You can see if there is a menu option to turn off power save or turn on always on, otherwise there is no fix.

hamdanfadi, 2021-03-15:
Thanks for your reply. You are right; however, it supports RTSP/NAS. It works with Synlogy DSM. You might need to look at this option in your future update(s).

Cheers,

starbasessd, 2021-03-15:
The difference being the camera doing the motion detection and then doing
the storage to a NAS (most IP cameras can do) versus
having motioneye/motioneyeOS doing the detection, manipulation, and storage
of a video stream.
motion/motionEye/motionEyeOS are dependant on that continuous stream. You
could ask the Motion Project to add a re-trigger or continuous monitor
feature. They are located at
https://github.com/Motion-Project/motion#resources

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

magicmb, 2021-08-20:
> The difference being the camera doing the motion detection and then doing the storage to a NAS (most IP cameras can do) versus having motioneye/motioneyeOS doing the detection, manipulation, and [...]

@starbasessd I also have the same issue. However, I've noticed  in the RTSP Stream configuration section in the Eufy App, which is used to setup NAS(RTSP) storage, there is an option to either choose 'Events' or 'Continuous Recording' in the 'Video type store to NAS' section.

It seems that Eufy cameras endorse the use of Surveillance Station on the NAS, but given that I have the 'Continuous Recording' option selected in the Eufy config and since I can view the live stream using VLC on my MacBook shouldn't I be able to set this RTSP stream link up in MotionEye running on say my Raspberry Pi 3B+ over wifi? 

Correct me if I'm wrong but if VLC is correctly displaying a continuous live stream I just don't get how MotionEye can't make use of this same live continuous stream.

Eufy Indoor Cam (2k)
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10

starbasessd, 2021-08-20:
What URL do you use in VLC?

magicmb, 2021-08-20:
> What URL do you use in VLC?

rtsp://user:pw@ip/live0

starbasessd, 2021-08-20:
And if you add a camera using:
rtsp://ip_address/live0
and put the user and password in the correct boxes, 
and try both the tcp and udp camera types?

magicmb, 2021-08-20:
That's how I have it set in MotionEye: rtsp://ip_address/live0 with both the Surveillance Username and Surveillance Password fields under the 'General Settings' section.

I don't see a field for camera type where I can choose tcp or udp. Where exactly is this located?

Oh BTW the only other changes I made was to use Streaming Port: 554 as listed on my eufy app instead of the default (which wasn't working anyway) and I also tried to changed the streaming frame rate to 15 after seeing a comment on the eufy forum from a chap who appeared to get it to work.

Streaming Port
--

<br class="Apple-interchange-newline">

Streaming Port
--

<br class="Apple-interchange-newline">

magicmb, 2021-08-20:
I probably should also have said that I'm accessing MotionEye via my browser using http://ip_address:8765/

starbasessd, 2021-08-20:
Ignore the actual URL (it's for a FOSCAM I'm testing)
Last Box "Camera" becomes a drop down with RTSP/TCP and RTSP/UDP
Try both. 
[image]

starbasessd, 2021-08-20:
Surveillance User and Password are for connecting to motionEye:8765 but only to watch video, unable to make changes, not to connect to other cameras.

magicmb, 2021-08-20:
> Ignore the actual URL (it's for a FOSCAM I'm testing)
> Last Box "Camera" becomes a drop down with RTSP/TCP and RTSP/UDP
> Try both.
> [image]

Ah! All right, didn't notice the UDP option in the dopdown. Tried this and got the same result.

magicmb, 2021-08-20:
> Surveillance User and Password are for connecting to motionEye:8765 but only to watch video, unable to make changes, not to connect to other cameras.

All right, but the Surveillance User and Password fields get automatically populated by the user/pw entries I provided on the add camera dialog.

starbasessd, 2021-08-20:
Then it may be that the camera and motionEye/motion/ffmpeg are just not compatible.

starbasessd, 2021-08-20:
Then something is broken. They are not related in any way.

magicmb, 2021-08-20:
OK you probably won't believe this, but I just DISABLED Authentication  mode on both the Eufy App and inside MotionEye and my first TCP camera entry stream just came up !

magicmb, 2021-08-20:
How bizarre! 

Now I wonder what the risk is of leaving the 'Authentication Mode: Disabled' instead of 'Basic' or 'Digest'. Does this seem odd to you ?

starbasessd, 2021-08-20:
No. I depend on other layers of security, and do not use cloud services for surveillance. That is why I use motionEye and motionEyeOS rather than d-link, or Foscam, or Nest, or Ring, or Google, or Amazon services.

magicmb, 2021-08-20:
> No. I depend on other layers of security, and do not use cloud services for surveillance. That is why I use motionEye and motionEyeOS rather than d-link, or Foscam, or Nest, or Ring, or Google, or Amazon services.

Hmmm ... I've been wondering about this myself lately. I always had this in mind when I looked for a camera that allowed RTSP streams for storage in addition to the SD card, but then the camera's app is kinda relying on their own cloud severs and this then makes me wonder how secure this setup is and if I can even use the camera without the app. My guess is probably not, not for these type of cameras; although now I'm wondering what will happen if I cut the internet link. Will the camera still stream the RTSP locally?

starbasessd, 2021-08-20:
You have to try it. 
BTW, I'm more concerned about 'privacy' than my firewalls and security failing. I don't want to give anyone outside my network a 'show' should I need to traipse through the house because the nearest towel in in the dryer in the laundry room, ever. To avoid that, I'd have to move/remove too many cameras.

magicmb, 2021-08-20:
> You have to try it.
> BTW, I'm more concerned about 'privacy' than my firewalls and security failing. I don't want to give anyone outside my network a 'show' should I need to traipse through the house [...]

Ahahahaha ... I hear you LOL and it's on my todo list since you just can't tell who has access to your data these days and if the camera app credentials are compromised so is the surveillance. Mind you a good precaution could be to remove the SD card altogether and limit it to live view mode and the RTSP stream.

Anyhow, thanks for all your helpful comments which are invaluable for peeps like me learning as I go along, much appreciated :)

starbasessd, 2021-08-20:
BTW, how do you know they don't 'store' on the cloud without your permission? Ever read the agreements?  My firewall actually blocks the d-link and Foscams from getting through to the internet for any reason. They have the ability to update firmware from a local file, they get time sync from my own time server, I can manage them locally, with a web interface, no need for a 'special app', etc. At least once a day, they get blocked from 'calling home' by my firewall. Some systems (Nest & Ring and others) REQUIRE them to talk to 'home' or they fail, hence I don't use them....

magicmb, 2021-08-20:
Indeed, this is what I'm not certain about. I suspect my eufy cam might fall into that category, but I need to figure out how to do what you did with blocking on the firewall. I only have the security that came with the router, but there must be some logs on there that might point me in the right direction.

starbasessd, 2021-08-20:
This is off-topic, but I use a business class router that has inbound and outbound firewall rules, and all my cameras are in a list that gets blocked from the internet. The whole surveillance systems are on their own vlans as well.

JamBalaya56562, 2026-11-10:
Closing, as this got no answer to which install and versions it was. A network camera that shows for a minute or two and then goes grey has its stream dropped; `motion.log` from that moment tells why. Please open a new issue with it if this still happens with a current release.

closed 2026-11-10 by JamBalaya56562