motioneye-project/motioneye #2064: USB Camera disappears, won't reattach after power cycles

isny123, 2021-03-13:
I have a small USB camera - it will sometimes attach and work perfectly.  Then, after a week or so, it will not show up, or show 'video device not detected' and the gray screen.  Sometimes a power cycle will fix it, reboot sometimes as well.  Sometimes nothing fixes it and I just give up.... it's a v4l device.  I had similar issues before with a different app (motion), so may be a v4l issue.

starbasessd, 2021-03-13:
I have a similar issue (won't discover after reboot). It's a Tiga Dashcam. It has a very small battery in it (SuperCap?) 
If I need to do a reboot, I actually have to power off for more than 30 seconds so the internal battery dies. If I do a normal reboot, no joy. Power off for >30 seconds, no problems. Happens everywhere I've used it. Pi, PC, Laptop...
Motion is part of motionEye. If you use something like RaspberryPiOS, and an app like Cheese, or VLC, does it happen there, too?

isny123, 2021-03-13:
I've had this issue since forever. However, to provide more info, it's an 808 keychain camera with a small battery in it as well.  30 seconds wasn't enough to get it to work, so trying a longer time.

Suzhou65, 2021-03-29:
I have a similar issue, My USB camera is Microsoft LifeCam Studio.

I was monitoring my ceiling leaking, so I setting it take auto snapshot every 10 minutes. But, I found that it will automatically turn off the camera it self, and I will lost all days snapshot (because it didn't turn off), great.

The fast way to reconnect the camera, without restart the system or reattached the power cable, just login. ```Disable the Video Device, save apply, wait a second, then enable it again```. This operate works in my case, maybe someones can try it.

Here is my log files
[messages.log](https://github.com/ccrisan/motioneye/files/6220142/messages.log)
[motion.log](https://github.com/ccrisan/motioneye/files/6220143/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneye/files/6220144/motioneye.log)

isny123, 2021-03-29:
That didn't work in my case.  I purchased a new camera and I haven't had the problem yet in a couple of weeks, so I'm thinking it's related to how the camera is handled.

starbasessd, 2021-03-29:
If a camera 'goes to sleep' ( usually for power saving) motion & motionEye
do not handle that well. They need a continuous video stream from the
camera. Since they all seem to do it differently, it's hard to know how to
wake every different kind up. Also, most users don't want to disable the
power save mode for various reasons.
Without the video stream motion/motionEye can't know if there is motion.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as this went away with a new camera, and the same happened with Motion alone, as said at the start. When a USB camera stops sending video or drops off the USB bus, for example to save power, Motion has nothing to read until the device comes back, and that is up to the camera and the kernel. Please open a new issue with `motion.log` and `dmesg` if a camera that stays connected is not picked up again with a current release.

closed 2026-11-08 by JamBalaya56562