motioneye-project/motioneye #2067: motioneye on DietPi - cannot logout or login anymore, no camera shown, site error plz refresh message

githlp, 2021-03-17:
I have an issue with dietpi user permanently logged in on dietpi-ip:8765 and no cam shown though i see the actual stream on dietpi-ip:8081..

i restartet motion.service, motioneye.service, the raspi .. nothing helps. When accessing :8765 the user DietPi is logged in (even on other browsers) and i have no clue how to logout and login as admin or as user again..

The "switch user" button only refreshes the site but the login dialog will not show up..

`sudo journalctl -u motioneye`

`DietPi meyectl[24360]: WARNING: 403 GET /login/?_=1614955804882&_username=_&_signature=bcda15e9cc244b885c24bb3730ba00a0b1b5e72f (192.168.100.10) 1.33m`

Is there another possibility to switch or logout current user vie different url or at the backend (docker-img) ?

starbasessd, 2021-03-17:
Have you cleared cache & cookies in you browsers?
Which Pi are you running it on?

githlp, 2021-03-17:
Yes i did.
G_DIETPI_VERSION_CORE=7, SUB=0, RC=2, master, MichaIng
Debian Buster
Linux DietPi 5.10.17-v7l+ #1403 SMP 
RPi 4 Model B (armv7l)

starbasessd, 2021-03-17:
Which install file did you download and install for DietPi? I see a half-dozen or so just for the Pi...
32 or 64 bit? One of the others? I don't want to have to go through all of them until I get those posted information...

githlp, 2021-03-17:
I installed it as "Optimized Package" within the "dietpi-software" application..

`head -20 /usr/local/lib/python2.7/dist-packages/motioneye-0.42.1.dist-info/METADATA`

shows

`Metadata-Version: 2.1
Name: motioneye
Version: 0.42.1
Summary: motionEye server
Home-page: https://github.com/ccrisan/motioneye/
Author: Calin Crisan
Author-email: ccrisan@gmail.com
License: GPLv3
Keywords: motion video surveillance frontend
Platform: UNKNOWN
Classifier: Development Status :: 4 - Beta
Classifier: Intended Audience :: End Users/Desktop
Classifier: Topic :: Multimedia :: Video
Classifier: License :: OSI Approved :: GNU Lesser General Public License v3 (LGPLv3)
Classifier: Programming Language :: Python :: 2
Classifier: Programming Language :: Python :: 2.7
Requires-Dist: tornado (<6,>=3.1)
Requires-Dist: jinja2
Requires-Dist: pillow
Requires-Dist: pycurl`

dietpi@DietPi:/$

starbasessd, 2021-03-17:
I was referring to DietPi itself, actually [grin] but I'm sure that would have been in my future questions...

starbasessd, 2021-03-17:
I downloaded the 32 bit version and it's currently updating.

githlp, 2021-03-17:
Yes, not easy to recover this information, but i installed the 32bit ARMv6-Buster.img ..

I ran it now for nearly a week and the login appears a little buggy to me since it does not always respond to the switch user button - now it won't show the login dialog anymore. That´s why i asked for an alternative way to logout a user..

starbasessd, 2021-03-17:
I don't know about the DietPi version, yet. For normal installs, clearing
cache & cookies does it, if you selected 'remember me'.
That is why I'm building one right now,

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-17:
It may be interaction with whatever changes DietPi made ot optimise the
install. The normal user for motionEye is 'admin' with no password.
I don't know if the modified install is 'normal', or a wrapper, or what,
yet.
Will update as I come across issues...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

githlp, 2021-03-17:
ok, thanks a lot!

I did not activate the 'remember me' function, also there should only exist the users "user" and "admin" for motioneye, no idea why it sticks on the user of the OS = "DietPi" ..

starbasessd, 2021-03-17:
If 'remember me' isn't turned on, normal versions prompt for username and
password every time, or allow you to switch user if it logs you in as user
with very minimal functionality.
First boot up after installing only motionEye I get
DietPi v7.0.2 : 14:29 - Wed 03/17/21
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - CPU temp : 51'C : 123'F (Running warm, but safe)
 - LAN IP : 192.168.2.xxx (eth0)
 - MOTD : Enjoy our new DietPi v7.0 <3
---
connect to http://[ip_address]:8765
received login box
signed in using admin and blank password, worked
added network mjpeg camera.
motionEye Version 0.42.1
Motion Version 4.1.1 << older version
OS Version Linux 5.4.83-v7l+

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

githlp, 2021-03-17:
Yes initially it worked and i was able to set passwords for admin and user ..

If it runs for a while and then you come back and open the admin web interface then all the sudden it will not show the login page anymore.. since the services run fine i have no clue how to fix that - i will install a third browser and try there ...

githlp, 2021-03-17:
ok, its not an issue of the browser.. does not work anymore on FF, Brave neither on chromium.

starbasessd, 2021-03-17:
rebooted.
Went back to login box on WebGUI. Logged in admin, blank
Normal admin access
Clicked switch user
username 'user' password blank
Normal low level access. no issues
Title Bar shows hostname, in this case DietPi (Not the username...)

starbasessd, 2021-03-17:
I would do a Backup (under Settings, General)
I would purge motioneye, reboot, and re-install motioneye.
Then test again.

starbasessd, 2021-03-17:
Which log?

githlp, 2021-03-17:
well, yes, but this happened already.. it is always after some time that this happens..

I had motioneye on Raspbian, on Raspibian-lite, Raspi OS and then i tried motioneyeos and now dietpi with motioneye..

So for me the question is: what does the "switch user" button really do..

githlp, 2021-03-17:
> Which log?

sudo journalctl -u motioneye

starbasessd, 2021-03-17:
Takes you back to the login box, AFAIK.
That error seems to be related to it going back to the login box.
Timestamps correlate to  my clicking switch user and it going back to the login box. 
The only log I see is for motion. syslog doesn't exist. dmesg doesn't exist. I'm going to have to research turning on normal system logs...

starbasessd, 2021-03-17:
Are you saying it happens in all the other environments, too? I've NEVER seen it reported in any other environment, especially with recent images.
I'm going to do the logging research, and also try to install using one of the Wiki methods (Debian 20.04 and newer first) into DietPi.

starbasessd, 2021-03-17:
Check in your browsers:
Saved Passwords for the DietPi box IP address. If there, delete it.
I set password for admin to admin, user to user, repeated tests, and still the same.
I do NOT select "Remember Me" and I select 'never save' for this site, and have no issues.
I cannot leave it up a week (I use that Pi4 for other things) for further tests.
The log on my production server for motioneye has the same entry when I click 'change user', so I don't believe it has anything to do with it, and it has been up for more than a month.

MichaIng, 2021-08-21:
Hey guys, landed here when trying to get the Python 3 branch running for the Bullseye images, where we're not supporting Python 2 anymore. At least on a VM, the V4L2 camera is not detected... different topic...

Do I get it right that the issue appears only when setting an admin user password? I cannot do that on the current Python 3 test build, as storing it fails. Will send a PR...

For the current Python 2 implementation, there is nothing special we do:
```
# Dependency installs: motion v4l-utils python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev zlib1g-dev
pip2 install -U motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
# setting "media_path /mnt/dietpi_userdata/motioneye"
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl start motioneye
```
On Raspberry Pi, the Pi camera module is enabled, respectively the extended firmware for the required GPU features. That's it, so basically exactly following the Wiki for Raspbian/Debian/Ubuntu instructions from the Wiki, with a changed media directory.

But actually I remember seen the described issue, not sure if I faced it myself or if it was a user report, but obviously it got solved somehow, at least I cannot find it anymore and didn't do a report here either. @starbasessd did you do further debugging?

starbasessd, 2021-08-21:
@michalng I have, but unable to reproduce. I can try again if needed. @githlp wasn't real responsive before when I was trying to get info to reproduce the issue.

MichaIng, 2021-08-21:
Ah sorry, then I misunderstood your test results on too quick reading. Okay so then we're both not able to reproduce, and at least I could not find any similar report on our GitHub or forum, which means it is a very rare/setup-specific issue. Related to Python 3 support, I'll likely run more tests in the near features, so I'll keep an eye on this, and subscribed to this issue anyway.

starbasessd, 2021-11-08:
Closing for lack of responses, unable to reproduce.

closed 2021-11-08 by starbasessd