motioneye-project/motioneye #2069: Invalid Content-Length: value

weyrleader, 2021-03-21:
I have been looking for a way to add my Kasa KC100 IP camera to motioneye, and found the following articles which looked to suggest it may be possible, in fact I actually used Josh Getters awesome home assistant addon and successfully got it working that way and pointed motioneye at the RTMP stream it creates from the Home Assistant server, however looking at his response in the second link below I thought it may be possible to configure motioneye to get the stream direct from the camera:

https://medium.com/@hu3vjeen/reverse-engineering-tp-link-kc100-bac4641bf1cd
https://github.com/softScheck/tplink-smartplug/issues/73#issuecomment-680984256

When adding the camera to motioneye, initially it complained about the certificate, but found a config option to ignore validating certificates, but now get a  "Invalid Content-Length: value" error. If you look at his curl command he has a flag to ignore Content-Length, so went looking to see if there was an option for motioneye to no avail. Is there such an option or is it possible to add this as a config option?

Regards
James

starbasessd, 2021-03-21:
> kc100-bac4641bf1cd
> softScheck/tplink-smartplug#73 (comment)

Did you try:  rtmp://ipaddress:43331/Live/BalconyCam

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

weyrleader, 2021-03-21:
That's the stream that gets created by running Josh Getters HA addon, that works fine but relies on running that addon to create the stream, and adds load to the HA server so I was looking to try and get the stream direct from the camera based on the URL in his response, with the base64 encoded the password, it looks to authenticate fine, just gets stuck on the returned content-length error so doesnt add the camera

starbasessd, 2021-03-21:
That is a direct from the camera stream to be manipulated by motion, and
should bypass need to use https and encoded passwords.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

weyrleader, 2021-03-21:
I assumed the same thing when I saw the entry on ispyconnect, but it didnt work. You'll notice the entry says this:

KC100 Home Assistant Integration | FFMPEG | rtmp:// | /live/BalconyCam
-- | -- | -- | --

Which lead me to find Josh's HA addon, when you set this up it creates the RTMP stream on your HA server so pointing HA at your HA servers IP with that URL does definitely work but its motioneye -> HA -> camera, not motioneye -> camera.

I have just manually created a camera.conf using this URL
https://ipaddress:19443/https/stream/mixed?video=h264&audio=g711&resolution=hd

with the email address used to create the KASA account the the base64 encoded password and after restarting motioneye it works, so this confirms motion doesn't require the content-length header, its just motioneye not creating the camera.conf entry because it sees the returned error when it connects.

weyrleader, 2021-03-21:
Correction.. the manual entry I created did not work. It looked like it did initially because I used the existing camera.conf that pointed at the HA stream as a template and just updated the netcam_userpass and netcam_url. As soon as I deleted the older entry the new one stopped working, and the logs are complaining that the https url is invalid, not exactly sure what I missed updating in the new conf file that somehow linked them @id maybe?

starbasessd, 2021-03-21:
I do not have a TP-Link KC100 camera to play with.
I've never had the URL generated from ispyconnect.com fail when the info is
entered correctly into the Add Cameras fields;
For example, if the camera IP address is 192.168.0.220, the username is
admin and the password is password, the URL generator gives:
***@***.***:43330/live/BalconyCam
***@***.***:43330/live/BalconyCam>*
*To translate to the fields for Add Camera:*
*URL: rtmp://192.168.0.220:43330/live/BalconyCam
<http://192.168.0.220:43330/live/BalconyCam>*
*Camera Type: Network Camera*
*Admin: admin*
*Password: password*
*Camera should auto-fill in*
Protocol: Since TCP is default, you should also try UDP
Do NOT try entering / editing the camera-X.conf files directly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as the KC100 sends its own stream over HTTPS, which Motion cannot read as an MJPEG or RTSP camera, hence the Content-Length error; the RTMP stream that the Home Assistant add-on makes from it worked, as found here. motionEye only passes the camera URL to Motion, so a camera that needs such a conversion still needs a tool like that add-on in between.

closed 2026-11-08 by JamBalaya56562