motioneye-project/motioneye #2070: Is there a filepath/filename length limit in the UI?

y8s, 2021-03-21:
I want to use the following for "image file name":
camname_%Y-%m-%d/camname_%Y-%m_%d_%I-%M-%S-%p_%q

and this for "movie file name":
camname_%Y-%m-%d/camname_%Y-%m_%d_%I-%M-%S-%p

but upon clicking the apply button, motioneye throws up the message
"Make sure all the configuration options are valid!"

When I look at the fields, they have the little red warning icon / mouseover text "special characters are not allowed in the filename", but I'm pretty sure I'm not using any because:
a) if I delete some from anywhere, eventually the warning vanishes
b) if I manually populate the filename in camera-1.conf, it works fine. files are saved with the right names in the right place.

Might be related to https://github.com/ccrisan/motioneye/issues/1497 but I dunno.

Thanks,
Matt

starbasessd, 2021-03-21:
Do you need 2 camname_'s in the filename? I use something like:
camname/jpg/%Y-%m-%d/%H-%M-%S-%q and camname/mp4/%Y-%m-%d/%H-%M-%S
or
camname/%Y-%m-%d/jpg/%H-%M-%S-q and camname/%Y-%m-%d/mp4/%H-%M-%S
to split them up.
Also, what value is %p? It's not in the tool tip, and may not be supported.

y8s, 2021-03-21:
I suppose I don't need two camnames, it just makes it easier to run scripts on multiple files, but that doesn't really answer my question.

%p is AM/PM. Here's some examples taken just now:

```
/.../camname_2021-03-21/camname_2021-03_21_11-57-14-AM_00.jpg
/.../camname_2021-03-21/camname_2021-03_21_11-57-22-AM_00.jpg
/.../camname_2021-03-21/camname_2021-03_21_12-06-14-PM_04.jpg
/.../camname_2021-03-21/camname_2021-03_21_12-06-16-PM_00.jpg
```

I edited my camname but you can see that the filenames work just fine.

starbasessd, 2021-03-21:
Unless your camera name includes special characters, I can't see any reason why a 60+ character path name (expanded) should cause a problem.

y8s, 2021-03-21:
That's the thing.. it's not causing a problem anywhere but in the motioneye interface. I can't apply any changes because it gives me the error above.

starbasessd, 2021-03-21:
I just did a copy/paste of your filenames above (literally camname in it)
Red warning.
If I change the %I for %H and remove the -%p it accepts it in the web interface. Since they are not mentioned in the WebGUI tool tip, there may be a filter for valid variables that isn't including those. I can tag as a request for 'enhancement' for the modification to allow them if you'd like.

y8s, 2021-03-21:
yes please. i like my hours <13.

starbasessd, 2021-03-21:
Tagged for 'enhancement'

zagrim, 2021-03-22:
The validator for filenames has this matcher as the part of it: `%[CYmdHMSqv]`
So, that's why `%l` and `%p` are not valid for the field. There's no real technical reason for that I guess (and the reason might be that those are not mentioned in Motion docs, either, they get inherited from `strftime` format specifiers). It would be trivial to add those two.

Myself I do find using 12 hour notation in file/path names terrible since it breaks up sorting by the file name (mixes day and night files), but then again I'm also used to 24 hour time notation.

starbasessd, 2021-03-22:
@zagrim many thanks for the clarification, as always.