motioneye-project/motioneye #2071: Missing Options (Admin Panel)?

DustyArmstrong, 2021-03-21:
Hi, 

Sorry if this is a silly question, but I seem to be missing the options to change hostname etc from the admin panel. I am logged in as administrator. I currently have no cameras set up - do I need to wait til I can add some (haven't arrived yet) or do further setup?

The only options I have are: 

```
Admin Username
Admin Password
Surveillance Username
Surveillance Password
```

I am running Motion Eye OS as a Docker container, motionEye version 0.42.1, Motion version 4.2.2 on Debian 9.12

Thanks.

starbasessd, 2021-03-21:
That is correct, you need at least 1 camera added before all the other admin functions are available.

DustyArmstrong, 2021-03-21:
Added a USB webcam I have laying around, and those options didn't show up.

starbasessd, 2021-03-21:
What does show up?
Which instructions did you use to install the focker image?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DustyArmstrong, 2021-03-21:
I get all the options for the camera, but no additional host options. So Video Device, File Storage, Streaming etc. 

I followed the resources on this repo, but also made some of my own modifications. 
```
version: "3.5"
services:
        motioneye:
                 image: ccrisan/motioneye:master-armhf
                 devices:
                         - /dev/video0:/dev/video0
                 ports:
                      - 8765:8765
                      - 8081:8081
                 volumes:
                         - /etc/localtime:/etc/localtime:ro
                         - ./meyevol/etc/motioneye:/etc/motioneye
                         - ./meyevol/var/lib/motioneye:/var/lib/motioneye
volumes:
        meyevol:

```

Docker works correctly, there isn't much to it and the camera works just fine, I'm just missing those particular options. Unless there's a different container for "server"?

starbasessd, 2021-03-21:
What is your docker install environment? Base OS docker is installed in?
Which Docker source? In my test environment (RPi4-4GB, RaspberryPiOS
Desktop, all current updates) I installed docker from sudo apt install
docker.io. I ran

docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf

And added a single camera. I then connected with a FireFox web browser
to the IP of the RPi
http://ip_address:8765
I get the login, I entered admin and get the Preferences and General Settings
I add a single camera, and I get all the settings available.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DustyArmstrong, 2021-03-21:
Docker installed on a base of Ubuntu server 20.04, on a Raspi4-4GB. 

Can you tell me what settings you see under Preferences and General Settings respectively? I have seen screenshots that show timezone and hostname in that General Settings area, but it's possible those just aren't included in the Docker version.

Either way I've found out how to modify the setting I wanted via the config file _motioneye.conf_. Edited the line:

```
server_name [yourserver]
```

starbasessd, 2021-03-21:
From my docker test install:
[image]

starbasessd, 2021-03-21:
These two sections option do not change whether there is a camera attached or not.

DustyArmstrong, 2021-03-21:
Thanks, that's exactly what I see too. I suppose those settings just aren't in the Docker image. Would be good if they were, but it's possible to modify what I wanted via the config so no great loss. 

Thanks for your help.

starbasessd, 2021-03-21:
The Hostname and Timezone are set when you run the docker, see my startup (and the primary startup listed in the Wiki Docs...

starbasessd, 2021-03-21:
--hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \

DustyArmstrong, 2021-03-21:
Thanks, I didn't see it in the example docker-compose file so I missed that. 

Either way, all sorted, thank you!

closed 2021-03-21 by DustyArmstrong