motioneye-project/motioneye #2072: High temp with rpi3

kemkem, 2021-03-23:
Hello

I'm currently using motioneyeos (2020606) inside a bird nest.
I can't access it easily since there's birds nesting in right now :(

I guess my raspberry 3 is not well refreshed, since the rpi is inside a platic box, inside a wood box..

temp measures :
```
vcgencmd measure_temp; vcgencmd get_throttled
temp=79.5'C
throttled=0x20000
```

It seems really high, and sometimes i can't access it anymore, I got to restart it (unplug / plug)

Is there a way to reduce temp ? I've already reduce cam resolution to 800x600/5fps, removed motion recording..

thanks

starbasessd, 2021-03-23:
Yes, throttled indicates it's forcing the ARM CPU cap, and yes, that is VERY warm. The normal way to cool is to add heat sinks and fans. You _might_ try a PiZeroW set to Fast Network Cam, or maybe an ESP32 IP web cam, and have them feed to the Pi3 outside the bird box...

starbasessd, 2021-03-23:
BTW, the proper github for motionEyeOS is:
https://github.com/ccrisan/motioneyeos/issues

JamBalaya56562, 2026-11-08:
Closing, as this was answered: the throttling shows the Pi 3 is too hot in its closed box, which needs heat sinks or a fan, or a smaller board such as a Pi Zero W as a network camera inside the nest, feeding a Pi outside, as starbasessd suggested. Lowering the resolution and frame rate, as already done, is what motionEye itself can do.

closed 2026-11-08 by JamBalaya56562