motioneye-project/motioneye #2073: Issue copying directory on Rasbian

waterbuffalo5, 2021-03-24:
Hi, I'm trying to install motioneye on Rasbian Lite, updated to 3/23/2021. I'm getting the error 
**cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory**. 

Pi Zero W

Motioneye seems to possibly have been collected, but not installed, as I can't find any directory relating to motioneye after **sudo pip install motioneye**.

I tried a fresh install twice.

Any help would be appreciated. 

Thanks

This code below isn't the original install, but running through the process a second time.
```
pi@raspberrypi:~ $ sudo apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12
Reading package lists... Done
Building dependency tree
Reading state information... Done
libmariadb3 is already the newest version (1:10.3.27-0+deb10u1).
libmicrohttpd12 is already the newest version (0.9.62-1).
libpq5 is already the newest version (11.10-0+deb10u1).
ffmpeg is already the newest version (7:4.1.6-1~deb10u1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ sudo wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
--2021-03-24 05:02:18--  https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
Resolving github.com (github.com)... 140.82.112.4
Connecting to github.com (github.com)|140.82.112.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-releases.githubusercontent.com/20822620/77fdcb80-bc14-11e9-81f5-70e7bfc5bdfb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210324%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210324T050219Z&X-Amz-Expires=300&X-Amz-Signature=d2b7ffd259460b91fc13504d2adbf54d49dccfe7822832449d299c801205a678&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=20822620&response-content-disposition=attachment%3B%20filename%3Dpi_buster_motion_4.2.2-1_armhf.deb&response-content-type=application%2Foctet-stream [following]
--2021-03-24 05:02:19--  https://github-releases.githubusercontent.com/20822620/77fdcb80-bc14-11e9-81f5-70e7bfc5bdfb?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210324%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210324T050219Z&X-Amz-Expires=300&X-Amz-Signature=d2b7ffd259460b91fc13504d2adbf54d49dccfe7822832449d299c801205a678&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=20822620&response-content-disposition=attachment%3B%20filename%3Dpi_buster_motion_4.2.2-1_armhf.deb&response-content-type=application%2Foctet-stream
Resolving github-releases.githubusercontent.com (github-releases.githubusercontent.com)... 185.199.111.154, 185.199.109.154, 185.199.108.154, ...
Connecting to github-releases.githubusercontent.com (github-releases.githubusercontent.com)|185.199.111.154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 312748 (305K) [application/octet-stream]
Saving to: ‘pi_buster_motion_4.2.2-1_armhf.deb’

pi_buster_motion_4.2.2-1_armhf.deb                          100%[========================================================================================================================================>] 305.42K  1.15MB/s    in 0.3s

2021-03-24 05:02:19 (1.15 MB/s) - ‘pi_buster_motion_4.2.2-1_armhf.deb’ saved [312748/312748]

pi@raspberrypi:~ $ sudo dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
(Reading database ... 42988 files and directories currently installed.)
Preparing to unpack pi_buster_motion_4.2.2-1_armhf.deb ...
Unpacking motion (4.2.2-1) over (4.2.2-1) ...
Setting up motion (4.2.2-1) ...
Processing triggers for systemd (241-7~deb10u6+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
pi@raspberrypi:~ $ sudo apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
libcurl4-openssl-dev is already the newest version (7.64.0-4+deb10u1).
libjpeg-dev is already the newest version (1:1.5.2-2+deb10u1).
python-dev is already the newest version (2.7.16-1).
zlib1g-dev is already the newest version (1:1.2.11.dfsg-1).
libssl-dev is already the newest version (1.1.1d-0+deb10u5+rpt2).
python-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspberrypi:~ $ sudo pip install motioneye
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting motioneye
  Using cached https://files.pythonhosted.org/packages/c5/ff/139b4b02ed089b2b50eee65d1059ae0d29064ab5f4ad429e45c702396697/motioneye-0.42.1.tar.gz
Collecting jinja2 (from motioneye)
  Using cached https://files.pythonhosted.org/packages/7e/c2/1eece8c95ddbc9b1aeb64f5783a9e07a286de42191b7204d67b7496ddf35/Jinja2-2.11.3-py2.py3-none-any.whl
Collecting pillow (from motioneye)
  Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf133452d4f6e0fe80de2df7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    99% |████████████████████████████████| 37.8MB 1.7MB/s eta 0:00:01Killed
pi@raspberrypi:~ $ sudo mkdir -p /etc/motioneye
pi@raspberrypi:~ $ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
pi@raspberrypi:~ $
```

starbasessd, 2021-03-24:
Install Raspberry PiOS 2021-01-11 Lite to 16GB SDCard with Win32DiskImager
Add wpa_supplicant.conf & ssh.txt to /boot partition
eject and install in PiZeroW
Boot
Receive IP Address
Connect to IP Address via ssh (PuTTY) on Win 10.
ignore 'normal setups'
sudo su
apt update && apt upgrade -y
53 pkgs to be upgraded -- completed
raspi-config
     set localization, password, resolution, etc
reboot
sudo su
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12
wget
https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
pip install motioneye
pillow install fails, need to run 2nd line:
repeat pip install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample
/etc/motioneye/motioneye.conf
(completed successfully)
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local
/etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
Connect to http://IP Address:8765, working

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

derpicknicker1, 2021-03-28:
I have the same problem As @waterbuffalo5 

```
pi@raspberrypi:~ $ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
cp: cannot stat '/usr/local/share/motioneye/extra/motioneye.conf.sample': No such file or directory
```

Any suggestions?

starbasessd, 2021-03-28:
Usually indicates python-pillow (and the following packages) failed to
install.

*note: If pillow installation fails (hangs and ends at 99%), you can
install it from official repos using apt-get install python-pil and rerun
pip install motioneye.*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

waterbuffalo5, 2021-03-29:
apt-get install python-pil **worked**. I see now that it was stuck at 99%. I didn't understand that the first time. Thanks!

closed 2021-03-29 by waterbuffalo5