motioneye-project/motioneye #2074: unable-to-open-video-device / 4l2_device_capability

adinovak0402, 2021-03-25:
Hi, I bought a new raspberry 4 (2G), I used motioneye on my previous raspberry 3B + without any problems, now I tried to download older .img files directly for raspberry 4, but I still have the same problem:
[image]
I tried looking here, and the most common mistake is, modprobe bcm2835-v4l2 , but I tried this, I'm attaching a log here:
`[2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[3:ml3:Camera3] [WRN] [ALL] mlp_retry: Retrying until successful connection with camera
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[3:ml3:Camera3] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[3:ml3:Camera3] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
[1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Camera2] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[1:ml1:Camera1] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[2:ml2:Camera2] [ERR] [VID] vid_start: V4L2 device failed to open
[2:ml2:Camera2] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[2:ml2:Camera2] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[2:ml2:Camera2] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
[3:ml3:Camera3] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
[3:ml3:Camera3] [ERR] [VID] vid_start: V4L2 device failed to open
[3:ml3:Camera3] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
[3:ml3:Camera3] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
[3:ml3:Camera3] [CRT] [NET] motion_init: Substream not available.  Image sizes not modulo 16.
`
I must also point out that the camera worked for me in normal raspbian, please for any help / advice

starbasessd, 2021-03-25:
Are you using RaspberryPiOS or something else for the base OS then adding motion/motionEye or are you running motionEyeOS?

adinovak0402, 2021-03-25:
I'm going on pure motionEyeOS

starbasessd, 2021-03-25:
This is the motionEye github, please close here and open in:
https://github.com/ccrisan/motioneyeos/issues

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

adinovak0402, 2021-03-25:
Ok

closed 2021-03-25 by adinovak0402