motioneye-project/motioneye #2075: Can't upload with gdrive when movie_extpipe is used

Barmatuhin, 2021-03-26:
I wish to use movie_extpipe option for fine encoding and try two different encoder options (found them here - https://motion-project.github.io/motion_config.html#movie_extpipe)
`movie_extpipe x264 - --input-res %wx%h --fps %fps --bitrate 2000 --preset ultrafast --quiet -o %f.mp4

movie_extpipe ffmpeg -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -i pipe:0 -vcodec libx264 -preset ultrafast -f mp4 %f.mp4`

in local directory files creates correctly, with mp4 extension, but in systemctl log i see this records
`Mar 26 14:51:56 hass meyectl[64909]:    ERROR: failed to create movie preview for /DoorCamera/2021-03-26/14-51-34: Command '[u>
Mar 26 14:51:58 hass meyectl[64909]:    ERROR: failed to upload file "/DoorCamera/2021-03-26/14-51-34" with service gdrive: [E>
Mar 26 14:51:58 hass meyectl[64909]: Traceback (most recent call last):
Mar 26 14:51:58 hass meyectl[64909]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 1103, i>
Mar 26 14:51:58 hass meyectl[64909]:     service.upload_file(target_dir, filename, camera_name)
Mar 26 14:51:58 hass meyectl[64909]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 57, in >
Mar 26 14:51:58 hass meyectl[64909]:     ctime = os.path.getctime(filename)
Mar 26 14:51:58 hass meyectl[64909]:   File "/usr/lib/python2.7/genericpath.py", line 72, in getctime
Mar 26 14:51:58 hass meyectl[64909]:     return os.stat(filename).st_ctime
Mar 26 14:51:58 hass meyectl[64909]: OSError: [Errno 2] No such file or directory: '/DoorCamera/2021-03-26/14-51-34'`

When i use build-in encoder options, all goes fine and system create thumb files and upload .mp4 files to Google Drive

starbasessd, 2021-03-26:
Does this also involve HomeAssistant or similar?

Barmatuhin, 2021-03-26:
> 
> 
> Does this also involve HomeAssistant or similar?

HomeAssistant Core is installed on the same host with motioneye.

starbasessd, 2021-03-26:
If you install your baseOS (RaspberryPiOS, Ubuntu, etc) then only install motion/motionEye, (use the appropriate install instructions here) do you have the same issue?

Barmatuhin, 2021-03-26:
> 
> 
> If you install your baseOS (RaspberryPiOS, Ubuntu, etc) then only install motion/motionEye, (use the appropriate install instructions here) do you have the same issue?

I have UP Squared hardware (https://up-board.org/upsquared/specifications/) and Ubuntu 20.04 LTS server on top of it. So on top of ubuntu installed HomeAssistan Core and MotionEye manual installation (by this instruction)  - https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Both of them behind nginx reverse proxy with ssl
But once again, when i use built-in encoding option (h254 for example), all goes fine. In logs i can found that at the end of the path/filename string it have .mp4 extension  ("/DoorCamera/2021-03-26/15-33-31.mp4") and uploadservice.py script works correct. But using built-in option i can't make fine tune (presets, threads, bitrate e.t.c) so i need movie_extpipe option to use.

starbasessd, 2021-03-26:
Please take HomeAssistant Core out of the picture. 
the 'hass' section of the error message concerns me:
Mar 26 14:51:58 hass meyectl[64909]: ctime = os.path.getctime(filename)

Barmatuhin, 2021-03-26:
Ok, remove it in my starter post. It's just a hostname....

starbasessd, 2021-03-26:
Please start over with a clean setup minus the HomeAssistant Core.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Barmatuhin, 2021-03-26:
not a desirable solution, because of a malfunction in the extra motion options to rearrange the entire system. Maybe someone will suggest another option ...

starbasessd, 2021-03-26:
If you have a USB Thumb Drive, you can image and boot to it instead of the
SDCard.
There is a time/effort value to also put in place, especially for someone
with little to no experience.
I'll step out and let someone else put in their $0.02 worth.
Good luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-03-26:
You can also check over at the HomeAssistant Community support site, or
over at the Motion-Project github for their support, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zagrim, 2021-03-27:
I've never used `movie_extpipe` myself, but seeing this in the log `No such file or directory: '/DoorCamera/2021-03-26/14-51-34'` makes me wonder if the issue is such that after the postprocessing the resulting file is written completely elsewhere in the file system. As I said, I've not used that option myself, so I have no experience whether the extpipe is run in-place (write the output file to the same directory as the original) or if one has to take care of the output path and filename in the extpipe command.

So what you could try first is to search the file system for the `.mp4` files that have been generated (or whether they really are being generated at all) and you find them somewhere try to modify the extpipe command so that the files would end up in the right place. For searching you might use something like `sudo find / -name "*.mp4"` on the command line (it will take a while to complete).

zagrim, 2021-03-27:
I do note that you stated "in local directory files creates correctly", but that seems to contradict the error. Are you saying that really there is a directory named identically to path shown in the error message, and that there are mp4 file(s) there?

Barmatuhin, 2021-03-27:
> 
> 
> I do note that you stated "in local directory files creates correctly", but that seems to contradict the error. Are you saying that really there is a directory named identically to path shown in the error message, and that there are mp4 file(s) there?

Yes, I can confirm that the files in both cases (using extpipe or built-in encoding) are created in the correct directory and with the correct extension (.mp4).
I described the problem from the point of view, that the script for loading the recording results works differently. Namely, it does not understand that it is necessary to look for a file with the name 15-33-31.mp4, but looks for 15-33-31 (without extension), if you use extpipe. Look at my first post log example.
As parameters for extpipe i used different encoding capabilities (x264 and ffmpeg). Just took examples from here https://motion-project.github.io/motion_config.html#movie_extpipe. The file name is passed as a variable %f 

> movie_extpipe x264 - --input-res %wx%h --fps %fps --bitrate 2000 --preset ultrafast --quiet **-o %f.mp4**

Barmatuhin, 2021-03-27:
UPD: if i remove ".mp4" at the end of extpipe string (movie_extpipe x264 - --input-res %wx%h --fps %fps --bitrate 2000 --preset ultrafast --quiet -o %f) file is created and uploaded to Google Drive, BUT without extension. So this test confirms that uploadservices.py looks for filename without extension.
And use expipe without extension not an option, files on Google not usable. Need extra work to rename them

zagrim, 2021-03-28:
Ok, changing the original file name definitely can break things since whatever actions occur after saving a file seem to get the file name/path as a parameter and that will reflect the original name/path.

However, I wonder why are the original files without an extension in the first place. What are the settings you use for "Movies" and/or what other settings there might be that have an effect on the file naming? You'd probably have no issues if the original files were just named `.mp4`, even if that was somehow faked by the file name setting.

Barmatuhin, 2021-03-29:
Once again, to clarity. I remove the extra options and use the built-in encoding capabilities.
The settings look like this 
[image]
Service status output looks like this

> root@hass:/DoorCamera/2021-03-29# systemctl status motioneye.service
> ● motioneye.service - motionEye Server
>      Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
>      Active: active (running) since Fri 2021-03-26 15:24:04 EET; 3 days ago
>    Main PID: 70492 (meyectl)
>       Tasks: 10 (limit: 9268)
>      Memory: 959.3M
>      CGroup: /system.slice/motioneye.service
>              ├─ 70492 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
>              ├─ 70522 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
>              └─148958 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
> 
> Mar 29 21:32:49 hass motion[148685]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
> Mar 29 21:32:49 hass motion[148685]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
> Mar 29 21:32:49 hass motion[148685]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
> Mar 29 21:32:49 hass motion[148685]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
> Mar 29 21:42:28 hass motion[148958]: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)

After motion in camera in destination folder i have few files

> root@hass:/DoorCamera/2021-03-29# ls -la
> total 502456
> drwxr-xr-x 2 root root      4096 Mar 29 21:34 .
> drwxr-xr-x 3 root root     20480 Mar 29 21:33 ..
> -rw-r--r-- 1 root root 342783340 Mar 29 21:33 21-33-29.mp4
> -rw-r--r-- 1 root root    230150 Mar 29 21:34 21-33-29.mp4.thumb
> -rw-r--r-- 1 root root 171405876 Mar 29 21:34 21-33-56.mp4
> -rw-r--r-- 1 root root     60000 Mar 29 21:34 21-33-56.mp4.thumb

and two of them, with .mp4 extension uploaded on Google Drive.
No new records in log file (log in info mode).
All work as designed! Perfect!

Than i turn off Movies switch and add two lines in Extra Motion Options section
(nothing extrodinary)
> movie_extpipe ffmpeg -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -i pipe:0 -vcodec libx264 -preset ultrafast -f mp4 %f.mp4
> movie_extpipe_use on
And config looks like this
[image]

Setting applied and motion in camera generated

Service status output looks like this

> root@hass:/DoorCamera/2021-03-29# systemctl status motioneye.service
> ● motioneye.service - motionEye Server
>      Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
>      Active: active (running) since Fri 2021-03-26 15:24:04 EET; 3 days ago
>    Main PID: 70492 (meyectl)
>       Tasks: 10 (limit: 9268)
>      Memory: 1.0G
>      CGroup: /system.slice/motioneye.service
>              ├─ 70492 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
>              ├─ 70522 /usr/bin/python2 /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
>              └─148958 /usr/bin/motion -n -c /etc/motioneye/motion.conf -d 5
> 
> Mar 29 21:49:50 hass meyectl[70522]:    ERROR: failed to create movie preview for /DoorCamera/2021-03-29/21-49-22: Command '[u'ffmp>
> Mar 29 21:49:50 hass meyectl[70522]:    ERROR: failed to upload file "/DoorCamera/2021-03-29/21-49-22" with service gdrive: [Errno >
> Mar 29 21:49:50 hass meyectl[70522]: Traceback (most recent call last):
> Mar 29 21:49:50 hass meyectl[70522]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 1103, in upl>
> Mar 29 21:49:50 hass meyectl[70522]:     service.upload_file(target_dir, filename, camera_name)
> Mar 29 21:49:50 hass meyectl[70522]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/uploadservices.py", line 57, in uploa>
> Mar 29 21:49:50 hass meyectl[70522]:     ctime = os.path.getctime(filename)
> Mar 29 21:49:50 hass meyectl[70522]:   File "/usr/lib/python2.7/genericpath.py", line 72, in getctime
> Mar 29 21:49:50 hass meyectl[70522]:     return os.stat(filename).st_ctime
> Mar 29 21:49:50 hass meyectl[70522]: OSError: [Errno 2] No such file or directory: '/DoorCamera/2021-03-29/21-49-22'
  
listing of destination directory :

> root@hass:/DoorCamera/2021-03-29# ls -la
> total 601532
> drwxr-xr-x 2 root root      4096 Mar 29 21:49 .
> drwxr-xr-x 3 root root     20480 Mar 29 21:33 ..
> -rw-r--r-- 1 root root 342783340 Mar 29 21:33 21-33-29.mp4
> -rw-r--r-- 1 root root    230150 Mar 29 21:34 21-33-29.mp4.thumb
> -rw-r--r-- 1 root root 171405876 Mar 29 21:34 21-33-56.mp4
> -rw-r--r-- 1 root root     60000 Mar 29 21:34 21-33-56.mp4.thumb
> -rw-r--r-- 1 root root  66025213 Mar 29 21:49 21-49-22.mp4
> -rw-r--r-- 1 root root  35422377 Mar 29 21:50 21-49-51.mp4

two new files with the same .mp4 extension, but without .thumb-s and no one of them on Google Drive

uploadservices.py in general (some included def inside it) try to found file without .mp4 extension 

> OSError: [Errno 2] No such file or directory: '/DoorCamera/2021-03-29/21-49-22'

I'm not very good at programming and in Python in particular, but it seems to me that the script refers to the standard library to determine the file name, but since there are at least two versions of Python (2.7 and 3.0), then there are problems ...

starbasessd, 2021-03-29:
I will leave this open here, but note: Until and unless you start over
clean, without HameAutomation Core installed, and test, I will not go any
further with trying to support you. Maybe someone else will.
(BTW, I know GDrive does work without HA Core installed. Their Community
and Author have a tendency to point fingers over here, saying they are but
a wrapper, but it DOES work without that product installed.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Barmatuhin, 2021-03-29:
> 
> I will leave this open here, but note: Until and unless you start over clean, without HameAutomation Core installed, and test, I will not go any further with trying to support you. Maybe someone else will. (BTW, I know GDrive does work without HA Core installed. Their Community and Author have a tendency to point fingers over here, saying they are but a wrapper, but it DOES work without that product installed.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Ok! You WIN.
I waste about 3h to reinstall host. Now i have the same result but with fresh clear ubuntu + motioneye installed accordin this instruction (https://github.com/ccrisan/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29)
If you need fresh and clean installation proofs, will be glad to provide them to you
[packagelist.txt](https://github.com/ccrisan/motioneye/files/6224778/packagelist.txt)

starbasessd, 2021-03-29:
No one 'wins'. 
I am a volunteer here, doing the best I can to support an awesome product.
I reproduce most environments for new issues, whether on all the models of Pi (except the Compute Modules), most of the OSs for motionEye, all the versions of motionEyeOS since 20190911, and I can reproduce most other OSs like Ubuntu, Mint, Fedora, and others, VMWare environments, VirtualBox environments, and Docker environments. I'm fairly successful at it, too.
The version of Python shouldn't be an issue if you are able to successfully install Python2.7, and pip2 from the instructions. There also may be a failure to install python-pillow (python-pil) but even that is covered in the instructions. I am sitting here with a RPi4-4GB with Ubuntu 20.04, and a X64 VM with Ubuntu 20.04 with motion/motionEye installed and working properly. (No HA Core or other software). GDrive upload works on them just fine. I still have issues with a Docker install, but that doesn't apply here.
Enjoy.

starbasessd, 2021-03-30:
Install Ubuntu Server 20.04 from:
ubuntu-20.04.1-preinstalled-server-armhf+raspi.img
using Win32DiskImager to 16GB MicrSDCard from MicroCenter
insert to Pi3B+ and boot
SSH to IP address, forced to change password, auto log off
SSH to IP address
sudo su
apt update && apt upgrade -y
apt-get install ssh curl motion ffmpeg v4l-utils -y
reboot
SSH to IP address
sudo su
apt-get install python2 -y
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
python2 get-pip.py
apt-get install libffi-dev libzbar-dev libzbar0 -y
apt-get install python2-dev libssl-dev libcurl4-openssl-dev libjpeg-dev -y
apt-get install python-pil -y
pip2 install motioneye
mkdir -p /etc/motioneye
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
Connect to WebGUI at http://ip_address:8765
Install IP Camera (doesn't make any difference, it's what I have available tonight)
Settings, File Storage
Enable Upload Media Files
Select Upload Service
Enter / for Location (testing only, adjust later)
Click Obtain Key (takes you to Google Logon, pick account, say Yes... Get Key, copy it.
Paste in Authorization Key Field
Click APPLY at the top
Click Test Service
I always get 'successful'. (except with Docker images, then I get a bad web page going to get a new key.)
Same basic process with any other Base OS.
Clean install works.

zagrim, 2021-03-30:
Thanks @Barmatuhin for the illustrative report. As I said I've never used `movie_extpipe` and didn't earlier realise the way it works (I hope I got it right this time): pass whatever video the camera provides to external `ffmpeg` to process without the need for the `ffmpeg` built-in to Motion to be used at all (and thus leaving any "Movie" settings not relevant in ME).

But I still think the issue boils down to that way the extension not being included in the file name ME is using. After all, the eventual real file name is formed by the last parameter passed to `ffmeg` within the piped command. You see, normally when you set the format under "Movies" to MP4 ME knows that the resulting files are `.mp4`. Now, there nothing it can know of that (trying to interpret whatever command comes after `movie_extpipe` is not an option :smiley: ).

Unfortunately I'm not able to come up with any workaround for that right now. This could be made to work by modifying the cloud upload feature to look for files starting with the known file name (so then it would pick also files with have an added extension), but that sounds like something that would probably lead to new bugs later on (uploading files that were never intended to be uploaded). The other option would be to implement built-in support for `movie_extpipe` so that ME then would again be able to know the resulting file name, but that would be quite a task (needing to have a way for the user to set all the `ffmpeg` options as they wish).

Sadly, I think you are the first user to try the combination of `movie_extpipe` and cloud upload. I don't think anybody anticipated how that combination would work (or not work)...

zagrim, 2021-03-30:
And for that matter, I do not think the way you've done the install really matters here.

Barmatuhin, 2021-03-30:
> Install Ubuntu Server 20.04 from:
> ubuntu-20.04.1-preinstalled-server-armhf+raspi.img
> using Win32DiskImager to 16GB MicrSDCard from MicroCenter
> insert to Pi3B+ and boot
> SSH to IP address, forced to [...]

Barmatuhin, 2021-03-30:
Thank you for understanding. And I certainly appreciate your work and time spent, including this collaboration. I didn't want to make disrespect to you, it's just a natural reaction to the offer to reinstall the system for any problem that cannot be solved quickly.
But I also understand that in order to exclude collateral influence, a pure experiment is needed.
To the point. My scenario is almost one to one like yours, the difference is that I install the Ubuntu 20.04LTS server for the x86 platform, my hardware is on Intel chip (https://up-board.org/upsquared/specifications/).
It looks like this:
Boot up from usb with the ubuntu distribution and go through the installation process (nothing extra - next, next...)
I do not install additional packages, everything is by default
setting up wifi using nmtui
setting up ssh using a public key
update & upgrade
Than, continue with motioneye instruction for Ubuntu 20.04
in the third paragraph of the instruction there was indeed a slightly outdated link for downloading get-pip.py, but it was recently fixed.
python and pip2 are installed without problems.
Further setting. First, I try the built-in encoding capabilities (H.264 (.mp4)) - activate the Movies section - everything is by default
In the File Storage section, select the folder that was configured according to the instructions. I activate my Google Drive subscription without any problems (line with successful verification).
In this version, everything works like a clock, only the encoding performance is not good (raw-stream is taken from the camera, and in my camera the raw-stream gives only 5 fps). Since there is no way to influence the preset encoding settings, i'm  forced to go for a solution with an extpipe
Of the changes, only two lines in the Extra Motion Options section and disabling the Movies section  
> 
> Install Ubuntu Server 20.04 from:
> ubuntu-20.04.1-preinstalled-server-armhf+raspi.img
> using Win32DiskImager to 16GB MicrSDCard from MicroCenter
>.....

Barmatuhin, 2021-03-30:
You have understood the problem exactly as it is.
I'll add a little more details ...
I've tried playing with the settings at the end of the ffmpeg line. If you remove .mp4 at the end of the line (leave -o% f), then at the output, in the folder with files, i get files with name 21-49-51 (hour-minute-second) without extension at the end, BUT the procedure for upload and generating thumb files works
I can conclude that the variable %f does not contain the file extension. And since the same variable is most likely used in auxiliary scripts (uploading files to remote storages such as Google, creating thumb files), while the file name is being generated in the extpipe using only the %f variable, everything works. But this by itself is not convenient, to have a file without extension (Google cannot show it, there is no association) and it is more difficult to work with it locally, additional actions are needed to rename.
As soon as I add the extension (-o %f.mp4), as it described in examples, then the chain breaks, since on one side the file was generated with extension, and on the other side (the unloading script) they substitute %f variable with filename without extension mp4.
I understand that the situation is not simple, and whether the developer will vryat will expand the capabilities of inline coding. 

> 
> Thanks @Barmatuhin for the illustrative report. As I said I've never used `movie_extpipe` and didn't earlier realise the way it works (I hope I got it right this time): pass whatever video the camera provides to external `ffmpeg` to process without the need for the `ffmpeg` built-in to Motion to be used at all (and thus leaving any "Movie" settings not relevant in ME).
> 
> But I still think the issue boils down to that way the extension not being included in the file name ME is using. After all, the eventual real file name is formed by the last parameter passed to `ffmeg` within the piped command. You see, normally when you set the format under "Movies" to MP4 ME knows that the resulting files are `.mp4`. Now, there nothing it can know of that (trying to interpret whatever command comes after `movie_extpipe` is not an option 😃 ).
> 
> Unfortunately I'm not able to come up with any workaround for that right now. This could be made to work by modifying the cloud upload feature to look for files starting with the known file name (so then it would pick also files with have an added extension), but that sounds like something that would probably lead to new bugs later on (uploading files that were never intended to be uploaded). The other option would be to implement built-in support for `movie_extpipe` so that ME then would again be able to know the resulting file name, but that would be quite a task (needing to have a way for the user to set all the `ffmpeg` options as they wish).
> 
> Sadly, I think you are the first user to try the combination of `movie_extpipe` and cloud upload. I don't think anybody anticipated how that combination would work (or not work)...

Barmatuhin, 2021-04-16:
So.... no solution from developers...
Ugly workaround. Edit relayevent.sh in /usr/local/lib/python2.7/dist-packages/motioneye/scripts and replace filename variable substitution like this
filename="$4.mp4"

closed 2021-04-16 by Barmatuhin