motioneye-project/motioneye #2076: An error occured. Refreshing is recommended.

daAlekseev, 2021-03-26:
Faced a problem when adding a camera, it gives the following not very informative error "An error occured. Refreshing is recommended."
Does not reflect video, only a picture with a crossed-out camera
I decided to go to the motion.log, where I saw the following "/usr/local/bin/motion: error while loading shared libraries: libavutil.so.55: cannot open shared object file: No such file or directory".
Then I checked the daemon.log and saw 
"Mar 26 18:02:13 raspberrypi meyectl[2840]: ERROR: motion not running, starting it
Mar 26 18:02:13 raspberrypi meyectl[2840]: ERROR: failed to start motion: motion failed to start
Mar 26 18:02:13 raspberrypi meyectl[2840]: Traceback (most recent call last):
Mar 26 18:02:13 raspberrypi meyectl[2840]: File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 325, in checker
Mar 26 18:02:13 raspberrypi meyectl[2840]: motionctl.start()
Mar 26 18:02:13 raspberrypi meyectl[2840]: File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 129, in start
Mar 26 18:02:13 raspberrypi meyectl[2840]: raise Exception('motion failed to start')
Mar 26 18:02:13 raspberrypi meyectl[2840]: Exception: motion failed to start
Mar 26 18:02:13 raspberrypi meyectl[2840]: WARNING: Connect error on fd 14: ECONNREFUSED
Mar 26 18:02:13 raspberrypi meyectl[2840]: WARNING: Connect error on fd 14: ECONNREFUSED
Mar 26 18:02:14 raspberrypi meyectl[2840]: WARNING: Connect error on fd 14: ECONNREFUSED"

I have already tried everything that is possible, but the problem is still not solved, I really hope for help!!

[image]
[image]

starbasessd, 2021-03-26:
Did you reboot?
motion version not showing in Settings, General.
What device? Pi Zero, 1, 2B, 3B, B+, 4B, 4B-1G, 4B-2G, 4B-4G, 4B-8G?
Which install instructions?
What camera?

daAlekseev, 2021-03-26:
> Did you reboot?
> motion version not showing in Settings, General.
> What device? Pi Zero, 1, 2B, 3B, B+, 4B, 4B-1G, 4B-2G, 4B-4G, 4B-8G?
> Which install instructions?
> What camera?

Yes, I rebooted the raspberry
Device - pi 3B+
Camera - Raspberry Pi Camera Rev 1.3 (Asia version)
I took the setup instructions from here https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

starbasessd, 2021-03-26:
How did you install camera? What mode? It should be MMAL.
Can you run 
vcgencmd get_camera
and post result?
Did/does motion version show after the reboot in Settings/General?

daAlekseev, 2021-03-26:
> How did you install camera? What mode? It should be MMAL.
> Can you run
> vcgencmd get_camera
> and post result?
> Did/does motion version show after the reboot in Settings/General?

When I add a camera I choose the following settings as in the photo
[image]

Vcgencmd get_camera returns "supported=1 detected=1" 
Motion version is not shown after reboot

starbasessd, 2021-03-26:
Delete any installed versions of the camera.
Add Camera, 
Select Local MMAL Camera

starbasessd, 2021-03-26:
It is possible that the motion installation is corrupted, also. You can try just re-installing via
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb

daAlekseev, 2021-03-26:
> It is possible that the motion installation is corrupted, also. You can try just re-installing via
> dpkg -i pi_buster_motion_4.2.2-1_armhf.deb

I delete the previously added camera and get the same warning again
[image]
i reinstalled got the following
[image]
just in case, restarted the raspberry and again the same warning

starbasessd, 2021-03-26:
Something is corrupted.
Please consider reinstalling from scratch.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

daAlekseev, 2021-03-26:
> Something is corrupted. Please consider reinstalling from scratch.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I'm a beginner in this environment, so sorry if my questions are too weird, but do I just have to go through the installation instructions again? (https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) or need to uninstall something- then or reset before reinstalling?

starbasessd, 2021-03-26:
The only bad question is the one you don't ask.
I would re-install from scratch.
Re-image to RaspberryPiOS 2021-01-11 and do all the updates
Then follow the install instructions, make sure you do the manual install
of python-pil (highlighted instruction) then do the pip install motioneye.
Then complete the rest.
Then try to install the camera as Local MMAL, not Local V4L2.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

daAlekseev, 2021-03-26:
> The only bad question is the one you don't ask. I would re-install from scratch. Re-image to RaspberryPiOS 2021-01-11 and do all the updates Then follow the install instructions, make sure you do the manual install of python-pil (highlighted instruction) then do the pip install motioneye. Then complete the rest. Then try to install the camera as Local MMAL, not Local V4L2. On Fri, Mar 26, 2021 at 12:59 PM daAlekseev ***@***.***> wrote:
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I have the Raspbian operating system installed, do I need to reinstall it from scratch, or did I not quite understand it correctly? If so, how can I do this, just I bought a card with the system already installed?

starbasessd, 2021-03-26:
Yes, and there are MANY tutorials for doing so, including:
https://www.raspberrypi.org/documentation/installation/installing-images/
https://www.youtube.com/results?search_query=install+raspberry+pi+os+Pi+3B

I have the Raspbian operating system installed, do I need to reinstall it

> from scratch, or did I not quite understand it correctly? If so, how can I
> do this, just I bought a card with the system already installed?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

daAlekseev, 2021-03-26:
> Yes, and there are MANY tutorials for doing so, including: https://www.raspberrypi.org/documentation/installation/installing-images/ https://www.youtube.com/results?search_query=install+raspberry+pi+os+Pi+3B I have the Raspbian operating system installed, do I need to reinstall it
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

Solved a problem! There was no need to reinstall the operating system, the daemon was not started, it was started manually and everything worked
Anyway, thanks a lot for your help

closed 2021-03-26 by starbasessd