motioneye-project/motioneye #2077: Running behind Apache proxy

sebastian-raubach, 2021-03-26:
I'm trying to run motioneye behind an Apache reverse proxy.
I've read quite a few issues on here as well as the wiki on nginx and Apache and I have the following configuration:

```
RewriteRule /doggycam$        /doggycam/               [R,L]
RewriteRule /doggycam/(.*)$   http://192.168.0.61/$1   [P,L]
```

This works relatively well. The web interface shows up and all the CSS and everything else loads fine as well.
The only issue is that the video itself doesn't show. Looking at the dev console's network tab, I can see a request to
`http://mydomain:8081/?_=1616781429979` while everything else correctly requests from `http://mydomain/doggycam/`.

Why is the video the only thing that doesn't seem to work correctly  and how can I fix it?

Suzhou65, 2021-03-29:
You can running the fully apache server, then setting the ProxyPass, it will be easier. 
```config
<VirtualHost *:80>
	Protocols http/1.1

	ServerName localhost
	ServerAdmin example@gmail.com
	DocumentRoot /var/www/html

	ErrorLog ${APACHE_LOG_DIR}/error.log
	CustomLog ${APACHE_LOG_DIR}/access.log combined

	ErrorDocument 403 /403.html
	ErrorDocument 404 /404.html
	ErrorDocument 500 /500.html
	ErrorDocument 503 /503.html
	
	ProxyPass /cam01/ http://10.0.1.52/
	ProxyPass /cam02/ http://10.0.1.53/
	ProxyPassReverse /cam01/ http://10.0.1.52/
	ProxyPassReverse /cam02/ http://10.0.1.53/

</VirtualHost>

<VirtualHost *:443>
	Protocols h2

	ServerName raspberry.example.com
	ServerAdmin example@gmail.com
	DocumentRoot /var/www/html

	ErrorLog ${APACHE_LOG_DIR}/error.log
	CustomLog ${APACHE_LOG_DIR}/access.log combined
	
	ErrorDocument 403 /403.html
	ErrorDocument 404 /404.html
	ErrorDocument 500 /500.html
	ErrorDocument 503 /503.html

	SSLEngine on
	SSLCertificateFile /etc/ssl/example.crt
	SSLCertificateKeyFile /etc/ssl/example.key
	SSLCACertificateFile /etc/ssl/example_ca.crt

	ProxyPass /cam01/ http://10.0.1.52/
	ProxyPass /cam02/ http://10.0.1.53/
	ProxyPassReverse /cam01/ http://10.0.1.52/
	ProxyPassReverse /cam02/ http://10.0.1.53/
</VirtualHost>
```
I use this configuration in my raspberry pi Apache server. so I can view the camera both LAN and WAN environment

sebastian-raubach, 2021-03-29:
That doesn't seem to change anything. Motioneye still requests from the top level on port 8081 rather than from `/doggycam/`  on port 80.

Suzhou65, 2021-03-29:
Oh, I omitted the port configuration, sorry.

I set the Export setting to port 80, including every motionEye based camera in my environment.
[image]

sebastian-raubach, 2021-03-29:
It's already running on port 80. As I said everything (including loading CSS and JS) works perfectly fine. The only thing not working is the actual video.
Here's what it looks like. URL is `http://192.168.0.44/doggycam/`, but the only request that fails is the one for the video which for some reason sends the request to `http://102.168.0.44:8081/`, so wrong port and also not using the `/doggycam` path.
[image]

JamBalaya56562, 2026-11-15:
@sebastian-raubach The request to `http://mydomain:8081/?_=…` comes from a camera added as "Simple MJPEG Camera": the browser loads that camera's URL directly, replacing only `127.0.0.1` with the page's host, so it bypasses the proxy. Adding it as a "Network Camera" instead makes motionEye pass the frames through its own port, and so through your proxy. Or give the Simple MJPEG camera a URL your proxy serves. Is that your setup?