motioneye-project/motioneye #2078: ubuntuu 12.4.2 server get-pip error

kaikus, 2021-03-27:
cant install motioneye. get at step 
python2 get-pip.py
of installation guideline the following error:

xxx@motioneye:~$ sudo python2 get-pip.py

Hi there!

The URL you are using to fetch this script has changed, and this one will no
longer work. Please use get-pip.py from the following URL instead:

    https://bootstrap.pypa.io/pip/2.7/get-pip.py

Sorry if this change causes any inconvenience for you!

We don't have a good mechanism to make more gradual changes here, and this
renaming is a part of an effort to make it easier to us to update these
scripts, when there's a pip release. It's also essential for improving how we
handle the `get-pip.py` scripts, when pip drops support for a Python minor
version.

There are no more renames/URL changes planned, and we don't expect that a need
would arise to do this again in the near future.

Thanks for understanding!

- Pradyun, on behalf of the volunteers who maintain pip.

starbasessd, 2021-03-27:
What instructions are you following?
Why Ubuntu 12.4.2? That is so unsupported by anyone. Way, WAY past EOL support.

kaikus, 2021-03-27:
Oh sorry, that was an typo, of course latest 20.4.2. Used the install instructions here.

starbasessd, 2021-03-27:
Wow, interesting, to say the least. The https://bootstrap.pypa.io/pip/2.7/get-pip.py is where the instructions I updated a while back pointed, to, but it just downloads the error message now. Will check with the maintainers.

starbasessd, 2021-03-27:
Now, curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py works fine. Weird. They might have been editing or updating something.

kaikus, 2021-03-28:
can confirm, now working

closed 2021-03-28 by kaikus