motioneye-project/motioneye #2079: Summer time

cuagn, 2021-03-29:
I've installed motionEye in order to automate a timelapse.
Yesterday was a change from wintertime to summertime (in E.U.)
My server (raspi) has been automatically updated without any problem.
However, the timestamp of the pictures (OSD) given by motionEye is still in winter.

Maybe I have a bad parametrization somewhere in motionEye.
Or there is a bug somewhere.

starbasessd, 2021-03-29:
motionEye on another OS or motionEyeOS?
If motionEye on another OS, you need to set up the proper timezone in that OS.
If motionEyeOS you need to set up the proper time zone in Settings, General

starbasessd, 2021-03-29:
If in a docker or similar, there's a setting in docker to carry over the current time zone.

cuagn, 2021-03-29:
I've installed motionEye on a raspi4 running 2021-01-11-raspios-buster-armhf-lite.
The RTSP flow of the camera is set to a minimum and every minute a still image is captured and stored in a daily directory.
With an OSD giving the time stamp on the picture.
There is now a difference of an hour between the picture and the filename of the picture (aaaammdd-hhmmss.jpg).
Remark : I do not have access to the system which is far away in the Alps mountains (not network).
I'm just receiving an email with the last picture every two hours.
Just to say that until I'll be able to connect "in situ" I'm fully blind.

starbasessd, 2021-03-29:
I do not know if you set localization (language, keyboard, time zone in raspi-config, or not. Raspberry pi os defaults to english-gb, and UTC, IIRC. It may be that UTC is correct for winter, but not correct for summer. You probably need to set to a time zone that adjusts for DST for summer.
Until you can access it at the OS level, you're kinda stuck, sorry.

starbasessd, 2021-03-29:
Is the timestamp generated by the PI or the RTSP camera? Forgot to ask.
Could be the same issue if the camera is set wrong.
For example, my RTSP camera has it's own, and I cannot turn it off:
[image]

cuagn, 2021-03-29:
Timezone has been changed for Europe/Paris. And raspi has the right time. When it send an email, all is ok. I receive it a few seconds after. 
Sent time (smtp), filename of last image, time of email received, all is coherent.
BTW I'v doubt : is the OSD the camera one or the motionEye one?
I may have done a mistake, but, as I'm using motionEye in order to rationalyze all cameras management (only one for the moment) I'm (quite) sure that the time is the one of motionEye.
[image]
I will come back after a visit "in situ" in a couple of weeks.

Should I close this issue or could it remains pending?

Thank you
Marc

cuagn, 2021-03-29:
Thank you.
I close it and will come back with the explaination (may be).