motioneye-project/motioneye #2080: MotionEye Video Error ... randomly sometimes...

wilbolinux, 2021-03-29:
[image]
[image]

Hi there ...

It happened again... sometimes i have same video error...
but the cam (RTSP) is OK;

My WorkAround:

if i go to the Motioneye Videosettings... and i change something like Framerate and hit 
the APPLY Button..... the Videos shows normally function.

Did you know this error ? 
the error unfortunately occurs only sometimes 

motionEye Version | 0.42.1
Motion Version | 4.1.1
OS Version | Debian 10

Greetings
   Wilbo

starbasessd, 2021-03-29:
You are using an older version of motion. Current in the install instructions is 4.2.2, latest is 4.3.2
Many new features and bug fixes since 4.1.1

wilbolinux, 2021-03-30:
Same error... :-/

I Upgraded Motion... and rebooted..

**motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Debian 10**

....not bette... any suggestions ?

starbasessd, 2021-03-30:
If you can get to the console, or SSH in, run the following commands after the problem occurs, but before you do anything to reset it:
vcgencmd get_throttled
and
vcgencmd measure_temp 
and post results here, please.

starbasessd, 2021-03-30:
BTW, if you are NOT running a Pi, let me know that, too.

wilbolinux, 2021-03-30:
Hi Starbasessd 

... your right... :-)

its not running on a pi.

its a x86_64  Board..:
Intel(R) Atom(TM) x5-Z8350-Prozessor (2 MB Cache, 1,44 GHz bis 1,92 GHz) 64-Bit-Architektur; 4 Cores...

Do you mean Thermal problems ? 
hmm... i take a look on htop.... 30 % CPU Rating...

starbasessd, 2021-03-30:
https://itsfoss.com/check-laptop-cpu-temperature-ubuntu/
Works for Debian, too.
I had to use sudo apt install lm-sensors
then run watch -n 2 sensors
to get mine (running about mid 60'C (higher than I like) on my Lenovo M75
I5 4GB Ubuntu 20.04
Heat is one of the bigger triggers of intermittent issues on any system,
then lack of drive space...
You said the camera works normally if you connect directly to it when
motionEye is having the video issue, correct?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JamBalaya56562, 2026-11-08:
Closing, as this was with motionEye 0.42.1 and Motion 4.3.2 and got no logs from when it happened. Saving the video settings restarts Motion, so the picture coming back after Apply means Motion's connection to the RTSP stream had stalled and did not recover by itself. If this still happens with a current motionEye and Motion, please open a new issue with `motion.log` from the time the error shows.

closed 2026-11-08 by JamBalaya56562