motioneye-project/motioneye #2081: How can I update sendmail.py and rebuild binaries?

mertemin, 2021-03-30:
I'd like make a few changes to `sendmail.py`. Is there a suggested way to do this? I think motioneye is installed via `pip`. So I am guessing I need to remove that, clone this repo, make changes I wnat and then rebuild and run. Am I right?

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606 - Running on a Raspberry Pi

starbasessd, 2021-03-30:
Not exactly. Apt, dpkg, etc are not installed in motionEyeOS, so you cannot remove and add packages quite that easily, not to mention the motionEyeOS is mostly set to Read-Only.
You can force it to be Read-Write using Debug Mode, or manually remounting partitions. Most of the python scripts are compiled into .pyc, so you'd have to decompile, modify, then recompile.
There is the ability to rebuild the whole kit and kaboodle with instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source

JamBalaya56562, 2026-11-08:
Closing, as this was answered for motionEyeOS 20200606, where the system is read-only and its Python files are compiled, so a change means building the image, from its own repository (motioneye-project/motioneyeos). With motionEye installed by pip, as today's releases are, nothing needs building: clone this repository, change `motioneye/sendmail.py`, install it with `pip install .` using the same pip that installed motionEye, and restart motionEye.

closed 2026-11-08 by JamBalaya56562