motioneye-project/motioneye #2082: Problem with recording files when motion is detected

daAlekseev, 2021-03-30:
I am using Raspberry Pi 3B+ with the Raspbian operating system installed. I started the motionEye in the browser and received a picture from the camera, but the movement is not detected and, accordingly, there are no video and photo recordings, so I decided to look at the state of the motion and saw the following errors. Read many forums but don't understand how to fix it. I really need help!!
[image]

starbasessd, 2021-03-30:
Which set of instructions did you follow to install?
Did you try to set it to run as a User instead of Root?

daAlekseev, 2021-03-30:
> Which set of instructions did you follow to install?
> Did you try to set it to run as a User instead of Root?

Set of installation instructions used the following (https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)
What exactly needs to be run as User instead of root?

daAlekseev, 2021-03-30:
> If you were trying to Run as a Normal User:
> https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User
> you can get those kinds of permissions errors. If you weren't something else is going on.
> Did you modify Settings, File Storage for anything?

I have not configured the User mode.
I only changed /etc/default/motion to start the daemon, since the camera picture was not displayed in motionEye

starbasessd, 2021-03-30:
Did you not run Step 7 in the install instructions?
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

Add an init script, configure it to run at startup and start the motionEye server:

 cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
 systemctl daemon-reload
 systemctl enable motioneye
 systemctl start motioneye

daAlekseev, 2021-03-30:
> Did you not run Step 7 in the install instructions?
> https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
>
> Add an init script, configure it to run at startup and start the motionEye server: [...]

I followed this point in the installation instructions, I meant that I did not try to work as a normal user

starbasessd, 2021-03-30:
What URL did you use to try to connect when you weren't able to get to
motionEye?
http://ip_address
or http://ip_address:8765?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

daAlekseev, 2021-03-30:
> What URL did you use to try to connect when you weren't able to get to motionEye? http://ip_address or http://ip_address:8765?
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

I connect by http://ip_address:8765 and I get a picture from the camera, but the movements are not recorded as I said earlier
 [image]

starbasessd, 2021-03-30:
Assuming Motion Detection is Enabled, 
Try lowering Frame Change Threshold to .1% 
Enable Show Frame Changes
Disable Auto Threshold Tuning
Enable Auto Noise Detection
Enable Despeckle Filter
Set Maximum Frame Change to 0 (zero)
Adjust Change Threshold upwards after it starts detecting movement.
These settings are highly dependent on what's in the frame, how much movement there is, how fast the movement is, etc.

daAlekseev, 2021-03-30:
> Assuming Motion Detection is Enabled,
> Try lowering Frame Change Threshold to .1%
> Enable Show Frame Changes
> Disable Auto Threshold Tuning
> Enable Auto Noise Detection
> Enable Despeckle Filter
> Set [...]

I changed the parameters to match what you wrote, but I get the same error
As far as I understand, motion is detected, but the problem is specifically with saving files or with something else
[image]

starbasessd, 2021-03-30:
Re: ERR Permission Denied

Sat Dec 09, 2017 1:54 pm
Edit motion config file:

sudo nano /etc/motion/motion.conf

and uncomment the following option:

#target_dir /tmp/motion

daAlekseev, 2021-03-30:
> Re: ERR Permission Denied
> 
> Sat Dec 09, 2017 1:54 pm
> Edit motion config file:
> 
> sudo nano /etc/motion/motion.conf
> 
> and uncomment the following option:
> 
> #target_dir /tmp/motion

In this file, I had the following line, and I uncommented it (but it is slightly different from what it should be, apparently)
[image]

then i checked the status of the motion and got it again

[image]

starbasessd, 2021-03-30:
I don't know. There is a file path where it's trying to save the file ffmpeg is trying to create, that doesn't have the correct permissions.
One of 2 possibilities:
Bad install 
or
something was changed, either accidentally or on purpose.
I just did a clean install (RaspberryPiOS 20210111 fully updated and install instructions from here) and it works fine.
I'm leaving it here for someone else to help if they can.

daAlekseev, 2021-03-30:
> Я не знаю. Существует путь к файлу, где он пытается сохранить файл, который пытается создать ffmpeg, который не имеет правильных разрешений.
> Одна из 2-х возможностей:
> Плохая установка
> или
> что-то изменилось-то ли случайно, то ли нарочно.
> Я только что сделал чистую установку (RaspberryPiOS 20210111 полностью обновлен и инструкции по установке отсюда), и она работает нормально.
> Я оставляю его здесь, чтобы кто-нибудь другой помог, если сможет.

Okay, then I'll try to reinstall the system. Now I have Raspbian installed. Is it better to reinstall it or install other system?

daAlekseev, 2021-03-30:
> I don't know. There is a file path where it's trying to save the file ffmpeg is trying to create, that doesn't have the correct permissions.
> One of 2 possibilities:
> Bad install
> or
> something was [...]

Okay, then I'll try to reinstall the system. Now I have Raspbian installed. Is it better to reinstall it or install a different system?

starbasessd, 2021-03-30:
On a Pi, I recommend motionEyeOS dev20201026, be sure to get the version
specifically for the Pi you have from here:
https://github.com/ccrisan/motioneyeos/releases
I runs only the bare minimum OS to allow motion/motionEye and is
specifically optimised to work well.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

daAlekseev, 2021-03-31:
> On a Pi, I recommend motionEyeOS dev20201026, be sure to get the version specifically for the Pi you have from here: https://github.com/ccrisan/motioneyeos/releases I runs only the bare minimum OS to allow motion/motionEye and is specifically optimised to work well.
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them what to do and let them surprise you with their results.” - G.S. Patton, Gen. USA Ethics are what we do when no one else is looking. Quis custodiet ipsos custodes? “There is no end to the good you can do if you don’t care who gets the credit.” - C Powell You know we're sitting on four million pounds of fuel, one nuclear weapon and a thing that has 270,000 moving parts built by the lowest bidder. Makes you feel good, doesn't it?

there is still a question, I decided to install a new raspbian, and so that errors did not occur again during the installation, I wanted to ask a question. Do I need to write these commands (sudo apt-get update
sudo apt-get upgrade) before this instruction (https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)?

starbasessd, 2021-03-31:
Absolutely, yes.

closed 2021-04-05 by daAlekseev