motioneye-project/motioneye #2083: raspberry pi + motioneye missing options in Camera Type menu

jivkottt, 2021-04-01:
I install motioneye on raspberry pi os but when i try to add network camera the option missing at the menu. also missing onother options. there is only two: remote motioneye camera and single MJPEG camera.
Can you help me to add another types? is it possible?
https://i.ibb.co/Zf022qf/IMG-20210331-200724.jpg

starbasessd, 2021-04-01:
You have a corrupted install. motion version is missing in Settings, General
Make sure you follow the steps in:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
It can hang at 99% when installing python-pillow, which means you need to use 
apt-get install python-pil 
and rerun pip install motioneye 
or start over (no need to fully re-image) with the install, and follow these steps:
sudo su
apt-get install ffmpeg libmariadb3 libpq5 libmicrohttpd12
wget https://github.com/Motion-Project/motion/releases/download/release-4.2.2/pi_buster_motion_4.2.2-1_armhf.deb
dpkg -i pi_buster_motion_4.2.2-1_armhf.deb
apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
apt-get install python-pil
pip install motioneye
mkdir -p /etc/motioneye
 cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
mkdir -p /var/lib/motioneye
cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
systemctl daemon-reload
systemctl enable motioneye
systemctl start motioneye
You should then be able to connect to the WebGUI, at port 8765. 
Go to Settings, General, and look for motion Version. It should be 4.2.2 if you followed the instructions above.
You should also be able to add all supported versions of cameras.

jivkottt, 2021-04-01:
thank you very mach! you are right. now all it's ok.
but I think will be better if you add in manual point 
3.1 apt-get install python-pil
https://ibb.co/gwCLVW3

starbasessd, 2021-04-01:
A lot of python users have python-pil/pyhon-pillow already installed, and the python-pillow instruction works for older versions of Raspbian (before RaspberryPiOS). I may create a second set like the Ubuntu 19.10 and Earlier vs 20.04 and later...

starbasessd, 2021-04-01:
But the missing motion install was your issue, not the pillow/pil issue.

JamBalaya56562, 2026-10-29:
As found above, the other camera types were missing because the install was incomplete (no motion version under General settings). The current install method in the README installs motion together with motionEye through `motioneye_init`: https://github.com/motioneye-project/motioneye#installation. Closing as completed.

closed 2026-10-29 by JamBalaya56562