motioneye-project/motioneye #2084: Call curl if no stream feed from camera is present

vedranius, 2021-04-01:
Hi,
I'd like to call this URL if no video stream from camera is coming into MotionEye:
http://192.168.1.11/cgi-bin/scripts?cmd=start&script=20-rtsp-server

Is that possible?

Tnx!

starbasessd, 2021-04-01:
Possible? Yes, I'm sure. I do not know where the change in screen display (showing grey screen or error screen) is called, though. Maybe one of the programmers knows.

zagrim, 2021-04-05:
I agree it would be nice to be able to have something like that to automate restarting of a flaky camera. But that might not be feasible, unfortunately. Possible, yes, but not very reliable or suitable for all users.

The error handler which displays the "no camera" image in place of the stream in case of error loading the stream is at [main.js#L4885](https://github.com/ccrisan/motioneye/blob/795616d15caa24fd35f1d41b37668086e1ce4049/motioneye/static/js/main.js#L4885). One could, theoretically, add there some kind of web request, but it would originate from the browser, not MotionEye server, and thus without creating some kind of backend endpoint the browser frontend could call in case of stream load error, the camera would need to be directly accessible from the browser (not true in at least my case, since I've firewalled the cameras pretty tightly). It might also be tricky to get it working correctly, so that constant restarting of the camera would not itself produce the condition which triggers restarting of the camera - I mean, the stream will not be available when the camera restarts. Anyway, personally I'd not propose adding any kind of error-request thingie in that place since it would architecturally be so wrong.

And I'm not quite sure if that would even cover all cases: I think also Motion will present some kind of error image in place of the stream in some cases.
The one controlled by MotionEye is the one which has camera icon with a diagonal line going over it (which is also shown briefly after making changes to camera config, and I think it corresponds to this kind of log entry: `WARNING: Connect error on fd 16: ECONNREFUSED`).
[image]

But then there's this kind of case:
[image]
That is something that is not made by MotionEye, and in that case I don't think ME has any idea that there actually is something wrong. From the look of the overlays I'd guess that is created by Motion.

zagrim, 2021-04-05:
The solution I'd propose here is to schedule a script on the MotionEye host to run, say, once every five minutes and somehow observe if the camera is still up and producing video stream (and if not, then call `curl`). I don't have alll the details of "observing if the camera is still up" thought out, but it could use the same network packet/byte counters utils like `nethogs` and `itop` use to display network usage by endpoints and/or what is available in `proc/<PID>/net` and/or try directly to listen to the camera stream and see if there's any significant amount of data received (now this would also tackle the "unable to open video device scenario"). It would take some scripting, but it would be doable.

JamBalaya56562, 2026-11-08:
Closing with a way that needs no polling: Motion runs the command in its `on_camera_lost` option when it loses a camera, and `on_camera_found` when the camera is back. motionEye passes such options through from Video Device > Extra Options, so a line like `on_camera_lost curl -s "http://192.168.1.11/cgi-bin/scripts?cmd=start&script=20-rtsp-server"` there calls the URL each time the stream is lost.

closed 2026-11-08 by JamBalaya56562