motioneye-project/motioneye #2085: the camera does not support newer versions of browsers

sargus123, 2021-04-02:
Camera | Received HTTP/0.9 when not allowed

the camera does not support browser versions newer than 44

  icsee application supports this camera.
    there is no update to this camera 

[image]

starbasessd, 2021-04-02:
Which camera?
Your command URL is saying 'command=snap' (for snapshot?) but motion/motionEye is looking for some sort of stream to manipulate...

sargus123, 2021-04-03:
I found a somewhat combined solution. I installed the iSpy application (windows 10).
and thanks to her I found the rstp address.
I got a tip from the page:   https://community.home-assistant.io/t/add-icsee-camera-to-ha-rtsp/290852 
to 10 points

starbasessd, 2021-04-03:
Would have helped here if you had posted the make and model of the camera here. I would have pointed you to the database for iSpy at https://www.ispyconnect.com/cameras
without the need to install any apps...

closed 2021-04-03 by starbasessd