motioneye-project/motioneye #2086: Connection timeout after some time (random: minutes to hours)

arielylim, 2021-04-03:
I'm having some issues with my motioneyeos installed on a Raspberry Pi Zero W. I got everything to work out properly—successfully flashed the OS, configure names, admin access, recording, and can even access from another device via SSH and FTP. 

My problem is after a while, which can be minutes or hours, I can't access it anymore. Not via SSH, FTP, or the IP address on browser. 

I keep getting a "connection timeout" or "can't connect to server" error. 

When i pull the plug and put it back on, it reloads, and now i can access everything again. I'm not sure how to troubleshoot it. 

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

arielylim, 2021-04-03:
Just happened again. As I was typing the first message, I just rebooted the Pi Zero. I accessed via browser to get the version, etc. 

Now, i get the connection timeout error. That's just about ~30 minutes after plugging it in. 

[image]

starbasessd, 2021-04-03:
This should be opened on the motionEyeOS github issues:
https://github.com/ccrisan/motioneyeos/issues
If you can, connect a monitor & keyboard to it. I would post the messages log at the very least (Settings, Expert) but all 5 logs may be of help. If you look at messages log, I'll bet there are thermal or voltage messages there.

arielylim, 2021-04-03:
Cool. Thanks. I didn't notice this was for motioneye only. I'll post there and include the logs.

starbasessd, 2021-04-03:
Closing in favor of 2714 in the motionEyeOS issues.

closed 2021-04-03 by starbasessd