motioneye-project/motioneye #2088: No browser video playback

jagauthier, 2021-04-05:
I'm running motioneye 0.42.1 on debian, and discovered that I can't play back the recorded videos in the browser.
There are simply no playback controls.  I understand that motioneye uses HTML5 video playback, so I tested again this site:
https://tekeye.uk/html/html5-video-test-page

All tests are fine.  Any suggestions?
[image]

starbasessd, 2021-04-05:
What format and location are you using to save files? At the least, I get a
listing of movies recorded, if I click the 3rd icon from the left ( the >
icon, next to the wrench)
I only get the pictures if I click the mountains with the dot (2nd from
left, next to the expand full screen)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2021-04-05:
Forgot: which browser and OS are you viewing from?

jagauthier, 2021-04-05:
It appears the movies are being stored as an AVI, I believe in /var/lib/motion/<camera>/<date>
[image]
Using Chrome Version 89.0.4389.114.
Have also tried with Edge Version 89.0.774.68

The attempts I've made (on multiple machines) have all been using Windows 10.

ozett, 2021-05-02:
the same at my side, but i made the timelapse with ffmepg on cli [with this command](https://github.com/ccrisan/motioneye/issues/1022#issuecomment-584016418)
but the resulting .avi dont play in gui.

( i remember years back i think that this once worked...)

[image]

jagauthier, 2021-05-02:
The issue is AVI. Change your video playback to mpeg.  That solved it for me,.

closed 2021-05-02 by jagauthier

ozett, 2021-05-02:
i tried this:

```ffmpeg -r 6 -f image2 -pattern_type glob -i './*/*.jpg' -c:v mjpeg -q:v 10 ./timelapse-02.mov```

what seems to me like mjpeg images to mjpeg-video, but also did not play in movies preview.
movie plays well in VLC ...

jagauthier, 2021-05-02:
What is this set to?
[image]

ozett, 2021-05-02:
in my case a made the movie with the command-line (see above), because i wanted a timelapse for more than one day.
so i dont have movies activated and only take still-images

[image]