motioneye-project/motioneye #2089: Sending notifications via Pushover

daAlekseev, 2021-04-05:
motionEye version 0.42.1
Motion version 4.2.2

There is the following misunderstanding. When notifications is sending about motion detection, I use the following script

#! / usr / bin / python
import sys, requests
r = requests.post ("https://api.pushover.net/1/messages.json", data = {"token": "token", "user": "user", "message": "Motion detected!"}, files = {" attachment ": open (sys.argv [1]," rb ")})

It works when called from terminal (python pushover.py <picture>) and notification comes
But when used in a motionEye nothing happens

kd7eir, 2024-11-21:
I know this is very old, but I see the same thing. Could you let me know if you found a solution?

JamBalaya56562, 2026-11-12:
The Run a command of Motion Notifications runs when an event starts, before any picture is saved, and gets no file path, so `sys.argv[1]` is missing in that script. To send each saved picture, put the command in File storage → Run a command, which gets the file's path as `%f`, e.g. `/usr/bin/python3 /path/to/pushover.py %f`. It runs as the user motionEye runs as, so the `requests` module must be installed for that Python. Closing as answered.

closed 2026-11-12 by JamBalaya56562