motioneye-project/motioneye #2090: Feature request: Define a timeframe for interval snapshots

ohjay93, 2021-04-06:
I would love to be able to set a timeframe for when to do Interval Snapshots.

Example:
Use the existing Interval Snapshots function, but limit it for a timeframe, i.e. from 09:00 to 18:00.

Use cases:
Create timelapses during a specific period, ie. during daylight, or just at night.
I use motioneye to monitor my plantbed, that gets UV daylight from 09:00 until midnight. After midnight the room is completely dark until next morning. During that period, taking a snapshot every 15 mins just fills up my memorycard with black photos I do not need, and which are cumbersome to delete before I make a timelapse.

Timelapses are both fun, beautiful and usefull to me; not only can I see how the buds grow, but I can also see if there are any organisms in one of my pots, that I need to take care of, so that they dont spread to my other plants.

starbasessd, 2021-04-06:
Also valid for motionEyeOS.

xjon685, 2021-04-07:
I have done this with a sonoff USB smart adaptor, that powers on my PI zerow for once a day for 10 mins and is set to interval snap shots every 360 seconds.   works well using another sbc as the motioneye server.    But I had always assumed that I could just set the interval snapshots to 43200 seconds and it will take a photo starting from when you first save the settings, but Ive not tested it.

starbasessd, 2021-04-07:
It'll take the snaps from the first time you set it, or from the next time it re-boots. Almost like a crontab with a @reboot command function.

ozett, 2021-05-02:
The requested feature ist great for not to take snapshots at certain time-periods/day-periods. And save diskspace.

I space is no problem, than the other requestes feature could come in handy, to make only the snapshots with certain images / from certain time-periods. https://github.com/ccrisan/motioneye/issues/2114

maybe helpful to have both options as a combined-feature?

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #883, which asks for the same: limit interval snapshots to a time window, like the working schedule does for motion detection. Making a timelapse from a chosen period is #2114.

closed 2026-11-13 by JamBalaya56562