motioneye-project/motioneye #2091: Transfering incomplete files due to cron job

xjon685, 2021-04-07:
motionEye Version | 0.42.1
Motion Version | 4.3.2
OS Version | Linux 5.4.83-v8  (docker / home assistant)

This is not an issue with Motioneye, as such more of a question.  If this is not the right place I apollogise.  

I have motioneye server running docker (home assistant) on a Rip 4,   all file uploads to FTP, SFTP Google or dropbox, fail to work, which seems to be a Home assistant issue rather that a motioneye issue.  But to get around the problem i havd setup my system as follows.

3 slave RPi 0w (motioneyeOS latest), 1 Pi 4 master (in Docker/HomeAssistant),  they save jpg and mp4 to a local Rpi 3 running openmediavault via SMB.  This all works fine,  from openmediavault, the files are sent foward to a remote server using rsync via a cron job on openmediavault. 

This all works well unless there is a file being written when the cron job executes, in which case the file is sent incomplete to the remote server.  I wanted to know if there is anyway to save the files with a diffent name until they comlete writing to the SMB attached storage.   That way rsync will see the files as separate files and won't ignore complete files that were partially transfered duing the last cron job.

I hope I made sense.

starbasessd, 2021-04-07:
It's an rsync issue, not motioneye or even OMV or HA.
Here's a good discussion, and a possible fix:
https://superuser.com/questions/847850/behavior-of-rsync-with-file-thats-still-being-written

xjon685, 2021-04-07:
I relise it's not a motioneye issue as I stated above, I was just asking if there is anyway  to save the files with a diffent name until they comlete writing that you know of?  

Anyway thanks for the suggestion and getting back to me.

starbasessd, 2021-04-07:
It is a limitation of rsync. There are lots of variables usable (like target_dir and such) that can be used in both motion.conf and motioneye.conf (in /etc/motioneye), but nothing that I am aware of to do that.
Could you call the rsync command from the Run an End Command (which runs after the detection and writing completes) instead of the cron rsync?

JamBalaya56562, 2026-11-08:
Closing, as the cause was found here: rsync from cron also picks up a movie that is still being written. As starbasessd suggested, the transfer can start when a file is complete instead: File Storage > Run a command runs after each movie ends and after each picture is saved, and Motion replaces `%f` in it with the file's path, so the command can copy just that file.

closed 2026-11-08 by JamBalaya56562