motioneye-project/motioneye #2093: Only 2 from 3 cameras record motion

hipitihop, 2021-04-09:
> apologies for hijacking #1857 I thought it was related.
@starbasessd apologies for the poor info there.

**Problem:**
I have been successfully running MotionEye on this setup with 2 x cameras for a few months. I then recently added a third camera. The new camera is identical model and firmware to one of the existing working cameras. 
MotionEye happily shows the stream in the webview like the other two working cameras which do record motion, but despite the same motion configuration settings, the new camera never captures/records motion.

Platform is a VM on Proxmox 6.3-6
OS Version | Ubuntu 18.10
motionEye Version | 0.42
Motion Version | 4.2.2

MoitionEye runs in docker via docker-compose as per https://github.com/ccrisan/motioneye/blob/dev/extra/docker-compose.yml

- my 3 cameras are all Ethernet/POE all on the same subnet.
- 2 x Ubiquitty Dome 3 + 1 x Beward Intercom.
- original ubiquity Cam2 works fine, second ubiquity Cam3 does not record motion.

----

Configurations:
**Cam1**
```
threshold_maximum 0
stream_quality 100
threshold 8595
noise_level 31
smart_mask_speed 0
pre_capture 125
movie_codec mpeg4
noise_tune on
stream_maxrate 30
netcam_url rtsp://192.168.1.99/av0_1
stream_localhost on
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 300
lightswitch_percent 80
movie_passthrough off
event_gap 15
auto_brightness off
stream_port 8081
rotate 0
stream_auth_method 0
threshold_tune on
framerate 25
emulate_motion off
movie_output on
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter EedDl
snapshot_interval 0
minimum_motion_frames 50
stream_motion off
target_dir /var/lib/motioneye/gates
movie_output_motion off
post_capture 250
stream_authentication **redacted**
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left gates
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
movie_quality 100
picture_output off
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t; /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc webhook -c /etc/motioneye/motioneye.conf 'POSTj' 'http://ha-nodered.pond:1880/gate-motion'
camera_name gates
netcam_tolerant_check on
height 960
mask_file /etc/motioneye/mask_1.pgm
width 1280
```

**Cam2**
```
threshold_maximum 0
stream_quality 100
threshold 1179
noise_level 1
smart_mask_speed 0
pre_capture 125
movie_codec mpeg4
noise_tune on
stream_maxrate 30
netcam_url rtsp://192.168.1.80:554/s2
stream_localhost on
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 300
lightswitch_percent 80
movie_passthrough off
event_gap 15
auto_brightness off
stream_port 8082
rotate 0
stream_auth_method 0
threshold_tune on
framerate 30
emulate_motion off
movie_output on
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter EedDl
snapshot_interval 0
minimum_motion_frames 125
stream_motion off
target_dir /var/lib/motioneye/unify1
movie_output_motion off
post_capture 250
stream_authentication **redacted**
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left unify1
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
movie_quality 75
picture_output off
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
camera_name unify1
netcam_tolerant_check on
height 576
mask_file /etc/motioneye/mask_2.pgm
width 1024
```

**Cam3** The one that does not record motion but otherwise works fine with the stream.
```
threshold_maximum 0
stream_quality 100
threshold 0
noise_level 31
smart_mask_speed 5
pre_capture 125
movie_codec mpeg4
noise_tune on
stream_maxrate 30
netcam_url rtsp://192.168.1.79:554/s2
stream_localhost on
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
movie_max_time 300
lightswitch_percent 70
movie_passthrough off
event_gap 15
auto_brightness off
stream_port 8083
rotate 0
stream_auth_method 0
threshold_tune on
framerate 30
emulate_motion off
movie_output on
picture_quality 85
snapshot_filename %Y-%m-%d/%H-%M-%S
despeckle_filter EedDl
snapshot_interval 0
minimum_motion_frames 25
stream_motion off
target_dir /var/lib/motioneye/unify2
movie_output_motion off
post_capture 250
stream_authentication **redacted**
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left unify2
picture_output_motion off
picture_filename %Y-%m-%d/%H-%M-%S
text_scale 1
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive off
movie_quality 75
picture_output off
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
camera_name unify2
netcam_tolerant_check on
height 576
mask_file 
width 1024
```

starbasessd, 2021-04-09:
A possible issue:
Cam 3 threshold set to 0 << Try 1179, same as Cam2
In the WebGUI I use 0.1%, it's hard to get it to select that vs 0.0

starbasessd, 2021-04-09:
BTW, Ubuntu 18.10 has been EOL'd since mid 2019 (it had an extremely short life cycle for some reason) and you may not be able to upgrade it very easily, but should consider for a production system to stick with the Even Year, .04 versions (LTS) 
It is at least 5 generations behind, and 4 versions of security updates alone.

zagrim, 2021-04-10:
I guess `threshold 0` might be due to having `threshold_tune on` (motion threshold is automatically adjusted, `threshold` setting ignored) and never setting it to any higher value. The same goes with `noise_level 31`and `noise_tune on` (I think 31 would be rather high for noise level).

Anyway, along the lines @starbasessd put it above, I'd suggest going with whatever settings you have on the 2nd cam (which detects motion). And if that does not help, use the standard manual method of tuning the motion related settings: switch all automation off (including smart mask), set motion threshold very low, 0.1% or so (I'm not sure if 0 for `threshold` somehow disables detection or not, at least Motion docs say valid values start from 1), and work from there until you only get "real" motion detected. If the scene is not controlled (say, outdoors), it might be tricky, I know.

hipitihop, 2021-04-12:
@starbasessd & @zagrim You were indeed right, as soon as I unticked the `threshold_tune` to expose the threshold slider and tweaked it to 0.1% , then re-enabled `threshold_tune` the camera started to detect motion. Many thanks for the extra pair of eyes. I will also schedule to upgrade my VM to the latest Ubuntu. Closing

closed 2021-04-12 by hipitihop