motioneye-project/motioneye #2094: Can't add motioneye camera to host in docker

arielylim, 2021-04-11:
I'm trying to add a motioneye camera in another device but i'm encountering an error saying no route to host. Not sure where to go from here. 

[image]

Here's the setup: 
- RPi4 running motioneye in docker. Doesn't have any camera installed
- RPi0W running motioneyeos with one camera installed
- I want to setup the RPi4 as my "single source of truth" where i can view the video the video from my RPi0W (and other future cameras i might add)
- I also want to set the recordings from all other cameras to my RPi4 later on.

My immediate problem is i can't add the camera from my RPi0 because of the error.

I have a feeling it has something to do with this section from the [docker install page](https://github.com/ccrisan/motioneye/wiki/Install-In-Docker), but i don't understand it. 

> ﻿﻿Add additional port mappings with the -p parameter if you want to use the streaming feature of motion: -p 8765:8765 -p 8081:8081; for cameras added, numbering of ports used for streaming starts from 8081 (second camera will use port 8082, etc). Ports used for streaming can be later changed in motionEye (Video Streaming -> Streaming Port) but should always match the ones that are being exposed from Docker.
> If using additional services that make use of ports in the 808x range, then default mapping can be edited to avoid conflicting by mapping higher range port numbers that are not in use by other services (i.e., -p 8765:8765 -p 58081:8081 -p 58082:8082)
>Change the two bind paths /etc/motioneye and /var/lib/motioneye according to your needs. The first contains the configuration files for motionEye and the second will be used as file storage for still images and movies. The bound file /etc/localtime is necessary for a proper timezone configuration inside the container using the timezone of the host.
>To forward a video device of your host use an additional parameter like the following
>  --device=/dev/video0

My setup follows the basic setup in the install page (but using portainer). 

I tried adding 8082:8082 but didn't work.

starbasessd, 2021-04-12:
Try for URL:
http://192.168.0.101:8081
assuming the IP address of the PiZeroW is 192.168.2.101
Confirm what the IP address of the Pi4 is, It should also begin 192.168.0.xxx
You should be able to ping the IP address of the PiZeroW from the Pi4.
Can you post the command you used to load and run the docker, please?

arielylim, 2021-04-12:
Tried using :8081 and got the same error. no route to host

Pi4: 192.168.0.100
Pi0: 192.169.0.101

Both are static IP assigned in router via mac address.

as for docker, i used portainer but i pretty much followed the instructions (i think).

volumes: 
- /etc/motioneye --> /etc/motioneye
- /var/lib/motioneye --> /var/lib/motioneye
- /etc/localtime --> /etc/localtime

ports: 
- 8081--> 8081/tcp
- 8081--> 8081/udp
- 8765-->8765/tcp 
- 8765-->8765/udp

starbasessd, 2021-04-12:
I haven't used portainer. I do use docker (in various flavors) and I know it works. Unfortunately, I don't think I can help you unless I know more about your Pi 4 setup.
What base OS, how/which portainer installed, and command(s) used to install motionEye. 
If you aren't using the docker commands given in the Wiki, are you doing a custom build? Does it include Python2.7 and the modified pip2 install, since both are beyond EOL and removed from most distros, now?

starbasessd, 2021-04-12:
I just made 3 attempts to use portainer to install motionEye, and failed.
I followed the instructions here:
https://pimylifeup.com/raspberry-pi-portainer/
and using the docker info from here:

docker run --name="motioneye" \
    -p 8765:8765 \
    --hostname="motioneye" \
    -v /etc/localtime:/etc/localtime:ro \
    -v /etc/motioneye:/etc/motioneye \
    -v /var/lib/motioneye:/var/lib/motioneye \
    --restart="always" \
    --detach=true \
    ccrisan/motioneye:master-armhf

It would pull down the image, but would fail to even start it.
I deleted the image after each attempt.
I then pasted the above command into the ssh session, and it immediately
started up and was functional.
I did notice a setting in Advanced, Network for Host or Bridged, and tried
both. neither successful
Your thoughts on what I may have missed, as I have not played with
portainer much.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

arielylim, 2021-04-24:
Sorry, i just had time to work on this now. 

Quick overview: 
- RPi4, running motioneye on docker
- RPi0, running motioneyeos
- I want to have my RPi4 serve as the main hub/server. Meaning, i want all recordings stored here. 

I am now encountering a new problem on my RPi4. 

```
pi@rpi4:~/motioneye $ docker-compose up -d
Removing motioneye_motioneye_1
Starting 54a398013b72_motioneye_motioneye_1 ... error

ERROR: for 54a398013b72_motioneye_motioneye_1  Cannot start service motioneye: error gathering device information while adding custom device "/dev/video0": no such file or directory

ERROR: for motioneye  Cannot start service motioneye: error gathering device information while adding custom device "/dev/video0": no such file or directory
ERROR: Encountered errors while bringing up the project.
```

which is weird because my `docker-compose.yml` file doesn't have the /dev/ anywhere. Could it be pulling from the image?

```
version: "3.5"

services:
  motioneye:
    image: ccrisan/motioneye:master-armhf
    ports:
      - "8081:8081"
      - "8765:8765"
    hostname: rpi4
    volumes:
      - '/etc/motioneye/config:/etc/motioneye'
      - '/var/lib/videos:/var/lib/motioneye'
      - '/etc/localtime:/etc/localtime:ro'
    restart: unless-stopped
```
anything i need to change in my compose file? 

Sidenote: not sure if this is important, but despite having the master-armhf as my image, when i ran the `docker-compose up -d`, i saw a download for the dev version. Couldn't find it anymore as i have rebooted several times already.

starbasessd, 2021-04-24:
As I stated, the motioneye container does not seem to play well with
portainer. On the same system, whether Pi, or PC, running it directly as a
docker command works fine. Nobody I've asked has seen a reason for it, but
they also have not been able to use the portainer wrapper. I'll leave it
up, with hopes someone else can assist.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

arielylim, 2021-04-24:
not using portainer anymore. That one is a direct `docker-compose.yml` similar to what is in the /extra folder.  

Let me try to re-install using the CLI

arielylim, 2021-04-24:
Finally got motioneye to work on my server. I was also able to add my RPi0 running motioneyeos on my server to be able to access in one browser. thanks for all the help. 

My last hurdle is how can i save the recordings from my camera (RPi0) to my server? 

[image]

Based on some preliminary research, i think i'm supposed to pick the network share. But then it goes on to talk about samba and setting up permissions in windows -- which isn't applicable. So, not sure how to proceed.

starbasessd, 2021-04-25:
What is the base OS on your Pi4, where you want to safe the files? 
Do you want to be able to access the files outside the docker container?

arielylim, 2021-04-25:
Base OS: Raspberry Pi OS

My RPi4 is using an SSD as a boot device.

And yes, I want to access it outside the container.

starbasessd, 2021-04-25:
I see two options:
1) add functionality to the RPi4  to enable SAMBA and or FTP services
2) Open a new volume in Docker, and use the Pi4 motioneye docker instance
to do all the heavy lifting for all the cameras.
I will start the instructions for #1 shortly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

aylim14, 2021-04-25:
Thanks! I tried the samba route. I got it to work between my Pi4 and Mac. But when I change the file storage to it on motioneye, I’m getting the error “Accessing network share failed: cannot create root directory”

I tried doing this with ufw disabled/enabled with samba allowed but both getting the same errors.

arielylim, 2021-04-26:
I'm not closing this issue. I found the solution [here](https://github.com/ccrisan/motioneyeos/issues/1652#issuecomment-789940411). Basically, the problem is that of a UI/UX, not in the code. 

For anyone who might encounter the same issue in the future, if you are using the **network share** share feature in the *file storage* area of motioneye, the key point to remember is that the Share Name **is not** the same as the folder/drive you named in your server (e.g. Samba). 

You have to specify a folder in it directly. See image below. 

[image]
  
So in my case, I was using the share folder as "rpi4share." I keep getting errors when i only point to that main folder. I created a folder inside it called "motioneye" and added another folder for "Camera1"

Enter your login credentials and keep the root directory as is. Hit Test Share to make sure it's working, then apply the settings.

closed 2021-04-26 by arielylim