motioneye-project/motioneye #2096: Notifications when start recording

ZoLuSs, 2021-04-13:
I received the notifications set in notifications script every time record video start, even when motion detection is OFF.

I have set a notifications script for notifying me when I'm out of office.
I've set 3600s per recording file.
Every time new file is created a notification is send.
Normally, notifications is sent only when motion detection is ON ?

starbasessd, 2021-04-13:
In order to receive a notification, 3 things need to be set PER CAMERA.
Motion Detection Enabled
Motion Notification, Send an Email enabled with valid email settings
Working Schedule, if enabled, matching time/date and condition.
Which version of motion, motionEye, and OS Version do you have? (Settings, General)

starbasessd, 2021-04-13:
If you are using a script, triggered by Run a Command, or Run an End Command, those are run anytime motion is detected and a file is created, regardless of the other settings, and if they email you when run, they will continue to do so, unless you are checking for time in the script and not sending during that time.

ZoLuSs, 2021-04-13:
I using:
motionEye Version	0.42.1
Motion Version	4.1.1
OS Version	Debian 10

So if I understand what you said, I can't use Continuous Recording and receive motion notifications with Run a Command ?

If I want to receive notifications with Run a Command I have to set Movies to Motion Triggered ?

starbasessd, 2021-04-13:
Correct. 
I would also upgrade Motion to at least 4.2.2, there were a lot of improvements and bug fixes from the 4.1.x versions. 4.3.2 is the latest (10/2020)

ZoLuSs, 2021-04-13:
Ok thanks for all !

closed 2021-04-13 by ZoLuSs