motioneye-project/motioneye #2098: Request - view for editing common settings across all cameras

will-mckeand, 2021-04-15:
I use motioneye with 6 RTSP cameras and while getting the hang of how all the settings worked I spent significant time copying settings from one camera to the remaining 5.

Would you be great is either an item in the cameras dropdown called 'Edit all' which displays and applies settings that apply to all configured cameras

OR

Next to the orange 'apply' button when changes are made, an 'apply to all' button which applies the modified settings to all cameras

Great work with motioneye - I've used it for a year and recently stepped up from an RPi 3 to an Intel i5 Ubuntu Server - oh the smooth video!

starbasessd, 2021-04-15:
Would cause problems for users with non-uniform cameras (me, for example) who would think the same setting would work for all cameras, but don't. If it applied all settings, to all cameras, it would break things on mine. I have 5 'permanent' cameras, a USB Webcam, 3 d-link dcs5020Ls, and a FosCam RTSP 9900. Almost none of the camera settings are valid across all. I use a couple of Pis with PiCams attached (CSI) and again, it would cause issues.

will-mckeand, 2021-04-15:
Thanks. That's true in your case but you would think most would have a collection of uniform cameras. I'm not a dev - just a cheapskate who didn't want to shell out for a mainsteam CCTV setup. Motioneye ticked all the boxes for me and this was my only niggle (that and apt installs an ancient version of motion which doesn't work with motioneye, that had me really puzzled!)

Thanks again

zagrim, 2021-04-16:
I almost could swear there has been talk about some kind of way to copy/clone cameras and/or their settings but I couldn't find a matching issue (maybe it has been a side-track in some issue).
I also do see the potential value for some users, especially when(re)building a setup with similar or identical cameras. I seem to remember that Shinobi had something like that, but just can't remember any details anymore. It might have been more for adding a new camera (thus anyway needing to review the settings) than copying settings from one cam to another.

Anyway, if it was done on per camera basis maybe it could still serve the purpose of reducing manual work in copying a bunch of settings and still allow to use it only on a subset of configured cameras. It should probably be restricted to settings that are sensible to copy accross different types of cameras, like "Text overlays", "Video streaming" (excluding the streaming port), "Still images", "Movies", "Motion detection" (well, those are likely to not work as-is on a camera with a different scene, but maybe some of it), "Motion notifications", and "Working schedule".

But if copying was only partial, maybe the UI should also somehow indicate which fields have been changed and which have not, so that the user would have at least some way of knowing what settings did change. That would be also a rather tedious thing to build probably, I don't think the current implementation of the UI fields uniformly supports changed/unchanged state...

JamBalaya56562, 2026-11-13:
Closing as a duplicate of #675, where @ccrisan suggested this very form: applying a setting across all, or some, of the cameras.

closed 2026-11-13 by JamBalaya56562