motioneye-project/motioneye #2100: Cannot add MotionEye Camera as a Network Camera using MJPEG streaming url

notcat, 2021-04-18:
Hello, I have an interesting setup with MotionEye installed on Raspbian OS, and a Debian Server that hosts a MotionEye instance that is the server for the client raspberry pis to connect to. I would like to use MotionEye as a streaming backend for the camera attached to the raspberry pis but I cannot.

When I try to add the Raspberry Pi's Streaming URL from MotionEye (running on the raspberry pi), https://192.168.24:8081/, it initially turns Green, then turns to Grey and says "Unable to connect to device"

It is really upsetting as I would normally connect it as a Remote MotionEye Camera but I cannot use it this way because when I try to use Motion Detection, the fps drops drastically from 30fps to 2-3. Using the Snapshot JPEG Streaming works, but the fps is capped at around 13-15.

I am willing to attach any and all files that could be helpful identifying the problem as I would be more than happy to help fix this issue for everyone.

starbasessd, 2021-04-18:
1) I hope the IP in the URL is just a typo.
2) Turn off Still Images, Movies, and Motion Detection on the Client Pis.
3) When you say a 'motioneye instance' what else are you running on the host? Are you running motionEye as a straight install, docker, or HA app, or something else? Which install instructions did you follow (URL)?
4) Does it initially allow you to add the camera as a Network Camera?
5) What are you running on the Pi Clients? Motion/motionEye, motionEyeOS? Which install instructions, or which version, on which Pis? Zero, 1, 2, 3B, 3B+, 4B 1GB, 2GB, 4GB, 8GB?
This is enough for a beginning.

notcat, 2021-04-18:
Thank you for the very extensive diagnostic procedure.

1. I am not sure what you mean by "a typo", When I click the Streaming URL it gives me that MJPEG stream.
2. The client Pi has nothing enabled except Streaming
3. I am running MotionEye as a straight install, following the instructions here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
4. It allowed me to add the streaming IP, it shows up as a MJPEG stream, and I confirm it, then it does the things I described earlier. Very green screen, no image, then loses connection to the camera.
5. I am running MotionEye on Raspberry Pi, GNU/Linux 10 (buster)  with nothing else installed other than a Wireguard server. I have a Raspberry Pi 4, 4GB version. I flashed using balenaEtcher 

Attached is a video of me adding the camera as a Network Camera.

https://user-images.githubusercontent.com/41898000/115154556-4ca9c180-a030-11eb-8c1f-60535ae5d596.mp4

starbasessd, 2021-04-18:
The typo is in the displayed link "https://192.168.24:8081/" only 3 octets shown
Are you adding the same camera 3 times to the same hub?
If you remove all 3 cameras, and ad them back, but add username 'admin' (even if blank password) does it make a difference?
I just fired up a VM test machine, and had no issue adding the same network mjpeg camera 3 times.
If it still doesn't work for you, I will take down a Pi4-8GB and recreate your setup.
By any chance, can you disable WireGuard for testing?

notcat, 2021-04-18:
I see what you mean in the first response, it was very late where I am when I posted that. Oops.

The 2nd camera is another raspberry pi camera, but 3b+ that I also was testing on. 

When I remove all cameras, add the Network Camera Streaming Url, nothing changes and gives me the same Green screen, into Gray, connection lost.

I disabled Wireguard and restarting the motion service and motioneye service, still no dice.
[image]

Attached is a video of me removing all cameras.
https://user-images.githubusercontent.com/41898000/115155472-f12e0280-a034-11eb-8ed0-ac158a006dc1.mp4

notcat, 2021-04-18:
Attached is a video of me removing all cameras.
https://user-images.githubusercontent.com/41898000/115155472-f12e0280-a034-11eb-8ed0-ac158a006dc1.mp4

starbasessd, 2021-04-18:
If you open another browser and go to 192.168.0.99:8081 what do you see?

notcat, 2021-04-18:
I see a regular MJPEG stream and it loads successfully, 

https://i.imgur.com/cX78BZl.jpeg

notcat, 2021-04-24:
Do you have any updates on this issue?

starbasessd, 2021-04-24:
I will be building a Debian 10 server (not Desktop) this weekend to test further.

notcat, 2021-05-12:
Hi was there any discovery in the same issue I am having with your test? Thanks.

starbasessd, 2021-05-13:
I started from scratch this morning. (again <sigh>)
Install Debian Buster Server on VM with 4GB/30GB, fully updated. 10.9.0-i386-netinstall
Installed motionEye for hub following instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Debian
Interesting. The only way  I am able to add a camera from another motionEye or motionEyeOS is as a Simple MJPEG Camera. 
I tried as Network Camera, Remote MotionEye Camera, and both failed with the green screen, then after a minute or two, would shift to grey screen. 
I added a password to "User" in case it was the lack of password. No Joy.
I upgraded to motion 4.3.2, and same issue. I will tag as 'bug' until / unless someone can point out what's not quite right.

tacocat75, 2022-07-21:
I have the same problem only I am using a pi3 as hub and my raspberry pi zero has motioneye .
both are using legacy raspberry pi OS (buster) and motion and motioneye are up to date.

To be clear they are both using motioneye not motioneyeos so no "fast network cam" feature.

The idea is to have the hub do the storage and motion detection, which cant be done as "remote motioneye"

If i go straight to the stream in my browser i can see the stream fine.

I had this working before set up as "network camera", If i remember correctly I got it to work by not using auth (or something like that) about 6 months ago, but I re purposed my pi zero for a while and now I get the green then grey screen mentioned above after setting it all back up again.

I have tried all the suggestions here and in this other thread here -
 https://groups.google.com/g/motioneye/c/-bWbOeBnTmw/m/a9Sm0whWAwAJ
https://groups.google.com/u/2/g/motioneye/c/D0t_ScFHxwI/m/AmfkRBKhAQAJ

Has anyone else had this issue lately or figured it out? it looks like it hasnt been discussed in a while.

I have tried all the suggestions here and in this other thread here -
 https://groups.google.com/g/motioneye/c/-bWbOeBnTmw/m/a9Sm0whWAwAJ
https://groups.google.com/u/2/g/motioneye/c/D0t_ScFHxwI/m/AmfkRBKhAQAJ

imagegrid, 2022-08-08:
I have motionEye 0.42.1 installed on Raspian Bullseye. Pi 4 as a hub with 2 Pi zeros, getting grey/green blank screens when adding the zeros as network cameras.  

Try adding the following to Extra Motion Options to the newly created Network Camera with the grey/green screen as mentioned in issue #1257. 

`netcam_keepalive=1 `

[image]

f-gdovin, 2023-06-18:
Had the exact same problem and adding `netcam_keepalive=1` fixed my issue. Just add that, hit Apply and voila, stream!. Thanks for this awesome software and for the support you provide.

fredca, 2023-06-22:
Thanks that solved my problem with a slightly different setup.
 Thanks for motion and motioneye they really do an excellent job.

JamBalaya56562, 2026-11-12:
I tried this on current versions: a Raspberry Pi 5 running motionEye on the dev branch with Motion 4.7.1, whose camera stream was added as a network camera to motionEye. Testing the URL in the Add Camera dialog detects keep-alive support, and the stream showed with keep-alive both off and on. As this was reported on 0.42.1 and does not happen on current versions, I'm closing it. If a motionEye stream still cannot be added, the workaround found above is keep-alive on: in current versions that is `keepalive = on` in the `netcam_params` line of the camera's `camera-N.conf`, which motionEye keeps when saving. Please open a new issue with Motion's log in that case.

closed 2026-11-12 by JamBalaya56562