motioneye-project/motioneye #2101: Ajout caméra MotionEye RPI zero dans MotionEye Docker

GBaud, 2021-04-18:
Bonjour à tous,
voilà des jours que je me rends fou : je tente en vain de rentrer une remote motioeye caméra dans le motioneye qui tourne dans un docker sur mon QNAP. 
Cette Caméra est installée sur un MotionEye qui tourne sur un RpiZéro à l'adresse : 192.168.0.61:8765, le port de stream vidéo est 8082. Elle fonctionne très bien. 
Dans le docker, je ne peux ajouter une caméra que si je mets l'adresse 192.168.0.61:8765, et que je paramètre les MotionEye sans mot de passe. Mais même en faisant cela, l'écran où devrait apparaître l'mage reste vide. 
J'ai tenté énormément d'adresses URL, mais sans succès. 

Je ne sais pas ce que je loupe... 
Si vous avez des idées... Je vous remercie par avance.

starbasessd, 2021-04-18:
Do you have multiple motionEye streams on the PiZeroW? If not, then the port you need is 8081, not 8082.

GBaud, 2021-04-18:
Thank you for your answer. I tried that too and it didn't change anything unfortunately. 
[image]

GBaud, 2021-04-18:
If it helps, here are my port redirects in the docker that should gather all the streams. 
[image]

starbasessd, 2021-04-18:
AFAIK (As Far As I Know) you can't have duplicate port forwarding in Docker. motionEye is TCP. RTSP cameras can be UDP.

GBaud, 2021-04-18:
I just changed the ports, and nothing changes. Still the black square with the search circle that rotates in the middle of the image.

starbasessd, 2021-04-18:
Unfortunately. beyond that, I don't know. I can and have used docker for motionEye on PiOS, Ubuntu, Debian, etc, with no issues. I run it for production on a M73 TFF with Ubuntu. I have lots of Pis I can toss a build on for testing, but no NAS (QNAP or other) that supports docker. 
I haven't had much luck with portainer, of HomeAssisstant, either.
I'll leave it here for others to see, maybe some one has experience. Google searching wasn't very helpful.

GBaud, 2021-04-18:
Thank you very much for your messages. I came to the same conclusions as you after searching on Google... 
On Qnap, the client called "container" looks a lot like Portainer indeed. 
I will try to put motionEye in a VM to see if the problem can come from this form of simulation...

GBaud, 2021-04-18:
I come back to this discussion for those who will come across this problem. 
I tested installing motioneye in a ubuntu VM and it worked right away. 
So I have two hypotheses: 
- either there was a port forwarding issue on the docker.
- either it's the containerisatoin under Debian that can cause more problems than Ubuntu. 

I can't check further than that for the moment. There will surely be other participants much better than me to correct my words.

starbasessd, 2021-04-18:
Works fine under Debian, and all it's derivatives like Ubuntu & RaspberryPiOS. The issues come when you use 'portainer', or HomeAssistant wrappers. Somewhat depressing that I can't get it to work there. I have even successfully built it from source with buildroot and thingos.

JamBalaya56562, 2026-11-10:
Closing, as the same remote camera worked right away from motionEye in an Ubuntu VM, which leaves the QNAP container setup as the cause, and this was with the 2021 image. Please open a new issue if a remote camera still stays black in a current Docker image.

closed 2026-11-10 by JamBalaya56562