motioneye-project/motioneye #2102: Connect Motion Eye camera in surveillance mode

tjerkkamstra, 2021-04-19:
Hi!

I have the following setup;
* MotionEye running in docker on my debian server
* Two Raspberry Pi's configured with motioneye attached to a Raspberry Pi webcam

This works perfectly. I can pull the images to the debian server no problem, and also added a spare phone running IP Webcam on android.

Now my friend also has a motioneye setup. He shared me the surveillance username and password and am able to login to his server and watch his cameras perfectly.

I would like to add his cameras to my server so I can see both my cams and his in one screen. I figured adding theirs as a remote motioneye camera would work, only it forces me to use the  'admin' username, which I do not have nor want the password for.

Would the only option be to ask him to stream his camera's and provide me with the surveillance credentials on another port? Or would it be possible to edit the username into something different then admin so you can add cams you have the surveillancedetails to as well?

Would also be interesting in a security setup: give a screen with view-only camera to the guards and set up a second instance to actually edit the camera setup.

Thanks in advance, I hope I explained myself sufficiently.

starbasessd, 2021-04-19:
I would like to add his cameras to my server so I can see both my cams and his in one screen. I figured adding theirs as a remote motioneye camera would work, only it forces me to use the 'admin' username, which I do not have nor want the password for.
     Use the 8081, 8082 ... ports of his server (assuming he's streaming as Network or Simple mjpeg Cameras). He'll have to forward them through the firewall, too.

Would also be interesting in a security setup: give a screen with view-only camera to the guards and set up a second instance to actually edit the camera setup.
     That is what the USER user is for.

If you need further help, just ask.

tjerkkamstra, 2021-04-19:
> Use the 8081, 8082 ... ports of his server (assuming he's streaming as Network or Simple mjpeg Cameras). He'll have to forward them through the firewall, too.

This was something I'd want to avoid, as he has to open up quite a few ports. I hoped you could add Remote MotionEye cameras using the surveillance user instead of the admin user and make use of the nice dropdown list of cameras.

> Would also be interesting in a security setup: give a screen with view-only camera to the guards and set up a second instance to actually edit the camera setup.
That is what the USER user is for.

Completely true. Forgive me my brainfart.

starbasessd, 2021-04-19:
This was something I'd want to avoid, as he has to open up quite a few ports. I hoped you could add Remote MotionEye cameras using the surveillance user instead of the admin user and make use of the nice dropdown list of cameras.
     Unfortunately, no, you can't. He could put the ports on a different outside port, like 48081, and set USER password (and actually change the USER name) and then add them with that name and password and only receive the stream.  The Username could be 12 random characters as could be the password. 
     If you are using motionEye, you would want to modify /etc/motioneye/motion.conf for the admin username and the user username.

JamBalaya56562, 2026-11-10:
Closing with the answer: a Remote motionEye Camera connects with the other server's Client Secret, which gives the hub access to that server's camera settings as well, so there is no view-only remote camera. With only the surveillance login, add the friend's cameras by their streaming URLs as Simple MJPEG Cameras instead.

closed 2026-11-10 by JamBalaya56562